ArduPilot/ardupilot #9843: PRX_ORIENT does not work (Teraranger EVO Tower)

GlidLov, 2018-11-16:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

### Bug report

**Issue details**

When setting PRX_ORIENT = 1 with a Teraranger Tower Evo hub board, mounted upside down, the sensors stop working and no data comes to proximity screen in MP. Otherwise, setting back to = 0 gives distance data.

**Version**
3.6.0

**Platform**
[  ] All
[  ] AntennaTracker
[ X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Hardware type**
Pixhawk 2.1 cube

auturgy, 2018-11-27:
I don't think anyone on the dev team has hardware to test with.  Perhaps @pl-kabaradjian can look into this?

GlidLov, 2018-12-03:
i can help do testing if needed

pl-kabaradjian, 2018-12-06:
Hello @GlidLov and @auturgy,

We tested in-house and indeed the Tower works correctly with PRX_ORIENT = 0 but not with PRX_ORIENT = 1. Unless we need to do something specific with this parameter in the driver, it looks like a bug in the ap_proximity library.

@rmackay9 any idea on what could be the issue?

IamPete1, 2021-09-07:
Looks like we just need to use the existing correction function in that backend.

https://github.com/ArduPilot/ardupilot/blob/35a93c598811fb153e0413dbcdb06b7aba6d1d47/libraries/AP_Proximity/AP_Proximity_Backend.cpp#L73-L78