ArduPilot/ardupilot #9849: Copter3.61 & 3.62RC2 F4BY startup fails before Baro carlibration when SD card is inserted.

Jasonsiu, 2018-11-18:
I have a problem on F4BY, it fails to boot as soon as a SD card is inserted. After switching back to Copter-3.60 the problem is gone. when switching to 3.61 and 3.62 the problem comes again. The startup Seemingly to before Baro carlibration. I checked SD card LOGS and TERRAIN folder on it. 3.61 and 3.62 could be startup Successfully until removed SD card .

Here is messages on MP
switching V3.6.2-rc2 SD card is inserted
Check BRD_TYPE: Baro: unable to initialise driver
Check BRD_TYPE: Baro: unable to initialise driver
Initialising APM
Check BRD_TYPE: Baro: unable to initialise driver
Initialising APM
Check BRD_TYPE: Baro: unable to initialise driver
Frame: QUAD
F4BY 0050004B 33375108 32343337
ChibiOS: ff603d11
ArduCopter V3.6.2-rc2 (f37cf570)

switching V3.6.1 SD card is inserted
Check BRD_TYPE: Baro: unable to initialise driver
Initialising APM
Check BRD_TYPE: Baro: unable to initialise driver
Check BRD_TYPE: Baro: unable to initialise driver
Initialising APM
Check BRD_TYPE: Baro: unable to initialise driver
Check BRD_TYPE: Baro: unable to initialise driver
Initialising APM
Frame: QUAD
F4BY 0050004B 33375108 32343337
ChibiOS: ff603d11
ArduCopter V3.6.1 (296132b8)
///////////////////////////////////////////////////

switching V3.6.2-rc2 card is removed
PreArm: Logging failed
EKF2 IMU0 tilt alignment complete
EKF2 IMU0 initial yaw alignment complete
Frame: QUAD
F4BY 004F004B 33375108 32343337
ChibiOS: ff603d11
ArduCopter V3.6.2-rc2 (f37cf570)

switching V3.6.1 card is removed
PreArm: Logging failed
EKF2 IMU0 tilt alignment complete
EKF2 IMU0 initial yaw alignment complete
Frame: QUAD
F4BY 004F004B 33375108 32343337
ChibiOS: ff603d11
ArduCopter V3.6.1 (296132b8)

///////////////////////////////////////////////////
switching V3.6.0 card is inserted
PreArm: RC Yaw not configured
PreArm: RC Throttle not configured
PreArm: RC Pitch not configured
PreArm: RC Roll not configured
PreArm: Compass not calibrated
PreArm: Compass not calibrated
PreArm: 3D Accel calibration needed
PreArm: Throttle below Failsafe
EKF2 IMU0 tilt alignment complete
EKF2 IMU0 initial yaw alignment complete
Frame: QUAD
F4BY 00230043 33375105 32343337
ChibiOS: ff603d11
ArduCopter V3.6.0 (2eedcf56)
No IO Thread Heartbeat ()
PreArm: RC Yaw not configured
PreArm: RC Throttle not configured
PreArm: RC Pitch not configured
PreArm: RC Roll not configured
Barometer calibration complete

kozinalexey, 2018-11-19:
Hi, I will check it with my controller, thank you for the information.

kozinalexey, 2018-11-19:
[image]

kozinalexey, 2018-11-19:
i have test with lastest build  2018 nov 19 8:49
http://firmware.ardupilot.org/Copter/latest/F4BY/
all looks fine

Try to erase parameters

what version of F4BY board you tested?

kozinalexey, 2018-11-19:
3.6.2 from beta builds work too http://firmware.ardupilot.org/Copter/beta/F4BY/ 
arducopter.apj 2018 nov 19 4:30
[image]

Jasonsiu, 2018-11-20:
@kozinalexey 
I tested on board 2.15(black), http://firmware.ardupilot.org/Copter/beta/F4BY/
arducopter.apj (arducopter 3.62 RC2) and http://firmware.ardupilot.org/Copter/stable/F4BY/
arducopter.apj (arducopter 3.61) . when SD card is inserted startup 3.62 fail. but removed SD card startup Successful. 
http://firmware.ardupilot.org/Copter/latest/F4BY/ 3.70-dev No problem was found. card  insert or remove FC still startuped Successful.
I saw your picture，logging failed in message, it look like haven't inserted SD card.  please insert SD card and try again to 3.61 and 3.62.

peterbarker, 2020-11-25:
Sounds like this was fixed long ago.... please reopen if not!

closed 2020-11-25 by peterbarker