ArduPilot/ardupilot #9861: KakuteF7 DMA on UARTs causes DShot issue

tridge, 2018-11-19:
This is a bug report from Paul Atkin @utkinpol 
Paul is using a KakuteF7, and finds that he needs to use NODMA on UARTs to get reliable flight on a small racing quad.
Paul, please put all details in this issue, including:

 - parameters (full file)
 - at least one DF log showing the issue
 - list of all peripherals (receiver, any uart connections, any i2c devices etc)
 - any other information that you think may be helpful

utkinpol, 2018-11-19:
Hi, 

Here are 2 logs with issue in them. 
https://drive.google.com/file/d/1on-t7zYsG4n-0A7huhVBmWKiX63Vbdwl/view?usp=sharing 4
https://drive.google.com/file/d/12A8e_RDZ0cZhyZ-hYMrs9ekvd4XKcdk7/view?usp=sharing 3

utkinpol, 2018-11-19:
[image]

this image shows current spike with no current reported by ESC

utkinpol, 2018-11-19:
the current fix of the problem was to apply NODMA to all UARTs and USARTs on the F7 chip. here is the parameters file
https://drive.google.com/file/d/1VVGtt0GrrpQOKBBUirm9iTxZX8LFrH0-/view?usp=sharing

utkinpol, 2018-11-19:
peripherals - esp8266 card, tfmini lidar, holibro gps, frsky r-xsr

utkinpol, 2018-11-21:
https://drive.google.com/file/d/1FT3bZgLd_jjss7yhwnY4XAbGhE_Ut_f9/view?usp=sharing

utkinpol, 2018-11-21:
issue is somewhat narrowed now. the zip file above has a good and bad log - a log '2018-11-21 12-55-36' has suffered a twitch on the motor A at mark 120 showing immediate jump of a current - it is easily seen. it has happened after testing a hwdef file with USART 1 and 2 given DMA channels. 

so, an issue here is - as it can be seen from the paramters, the USART1 has -1 option set, so it is not used. USART2 is used by the lidar, option 9 - and has minimal data flow rate. so it is unreasonable to suspect a DMA bus overflow. it is also odd to see how it would be a shortage of DMA channels, as only 2 usarts were added to dma - but, i am not positive about that, as something has happened. and it affected specifically motor A, pin PB1, associated with a DMA 1 stream 7.

utkinpol, 2018-11-21:
# USART1
PA10 USART1_RX USART1 
PA9  USART1_TX USART1 

# USART2
PD5 USART2_TX USART2 
PD6 USART2_RX USART2

utkinpol, 2018-11-21:
above is the configuration change where NODMA was removed. i searched through whole stm32f745ie.pdf doc to see where 7th channel of DMA1 could possible interfere with assignments of those pins or those USARTs - i found nothing.

chobitsfan, 2020-08-07:
Hi @utkinpol Would you mind taking a look at #14995 ? Thank you very much

PR #15256 (2020-09-08): AP_HAL_ChibiOS: fix kakutef7/mini motor glitch when using dshot
fix #9861
replace #15241 and #14995
closed 2020-09-08 by peterbarker