ArduPilot/ardupilot #9868: Copter is departing with wrong yaw on first switch to AUTO

hsteinhaus, 2018-11-20:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

### Bug report

**Issue details**

_Please describe the problem_

**Version**
_What version was the issue encountered with_

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_

**Logs**
_Please provide a link to any relevant logs that show the issue_

rmackay9, 2018-12-10:
I tried to reproduce this in SITL but couldn't immediately.  Of course that doesn't mean there isn't a problem just that I couldn't reproduce it.  The logs do look like there's a problem.  I would have expected ATT.DesYaw, ATT.Yaw and GPS.GCrs to all line up as long as the pilot hasn't provided any yaw input.  For much of the log this is true but it's not around lines 425000 to 550000.

hsteinhaus, 2018-12-10:
I think I have hundreds of similar logs on six different platforms (px4-v2,3,4 and fmuv2,3,4), the reproducability on real hardware is IMHO 100% since at least Copter 3.4.

Did you enter AUTO from PosHold?

hsteinhaus, 2019-04-01:
@rmackay9 It seems to be important to come from PosHold to corrupt the departure direction. When coming from Loiter, there is no such effect even in real world. From PosHold, I can reproduce it by a chance of nearly 100%.

rmackay9, 2019-04-01:
@hsteinhaus, ok, txs for the update

IamPete1, 2021-08-29:
I cant replicate when switching form PosHold, I suspect that this has been fixed with the change to Scurves. @hsteinhaus any update? Have you seen this on 4.1?

IamPete1, 2021-09-11:
Please re-open if this is still a issue.

closed 2021-09-11 by IamPete1