ArduPilot/ardupilot #9869: Copter: Enable 16 channels for Pixhawk/SBUS protocol

Seeelefant, 2018-11-20:
The 12 channel limit is really hitting me now,  especially when using a gimbal.  Technicallly SBUS is able to transmit 16 channels, Pixhawk offers 14 output  is there a way to enable these?

If not this is a pretty hot feature request

### Feature request

**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._

**Platform**
[  ] All
[  ] AntennaTracker
[X ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
_Add any other context or screenshots about the feature request here._

magicrub, 2018-11-21:
@Seeelefant what version are you running? I think we support 16 in and 16 out already

Seeelefant, 2018-11-21:
Hi Tom,
I am running the latest version ArduCopter V3.6.1 (296132b8) and at leat in Mission Planner I can only see 12 channels:

[image]

Please advise.
Best 
--Seeelefant

Seeelefant, 2018-11-21:
Would be highly appreciated, would be a big step for everybody using cameras . Drop me a line if I can help somehow.
By the way:  Is there a published road map somewhere out in the web?
Best
--Seeelefant

khancyr, 2018-11-21:
here : http://ardupilot.org/dev/docs/roadmap.html
But that is really generic, and there aren't point on the overall refactoring that we have been doing ...

dposea, 2018-12-10:
I have submitted a pull request but I think it's on hold until I also add plane and rover,  and I have been having weird compile issue that I'm sure I caused. New laptop. I'll get that done asap.

Seeelefant, 2019-01-04:
Hi dposea, I´ve seen that you made some progress on this topic. Do you have a timeline for it?

blacknose2010, 2020-10-16:
What is the Status on this ? 
I am seeing only 14 SBUS channels being output on a cube black on ArduPlane. 
(Was this ever planned to be put in plane too ?)
In my application I unavoidably need SBUS output ch 15+16 working...

andyp1per, 2020-10-16:
Should work on 4.1-dev

blacknose2010, 2020-10-16:
will try !

blacknose2010, 2020-10-16:
> Should work on 4.1-dev

unfortunately no change in behavior after flashing plane 4.1-dev firmware....
Parameter list is attached

[SBUS_ch15+16_param_.txt](https://github.com/ArduPilot/ardupilot/files/5392412/SBUS_ch15%2B16_param_.txt)

blacknose2010, 2020-10-16:
I see my problem now: my application is expecting ch15+16 to be digital SBUS channels as they would arrive from a RC rx.
Is there a way to get ardupilot output digital channels on SBUS ?

andyp1per, 2020-10-16:
Connect to a UART and use SERIALn_PROTOCOL 23  instead of RCIN/SBUS

blacknose2010, 2020-10-16:
Thanks for your reply andy, but dont get me wrong: the digital SBUS inputs are being _read_ correctly, just the SBUS outputs are not digital. I think this is a feature request, as I do not see a parameter that defines the style of SBUS output. (digital or proportional). 
I am asking for a digital sbus output being generated as described in Byte[23] in the SBUS frame. https://github.com/bolderflight/SBUS#description

IamPete1, 2021-08-29:
It sounds like this has been fixed, @blacknose2010 you use case is quite different to the original issue i think, if your still interested maybe you can open a new issue and provide some more information.

closed 2021-08-29 by IamPete1