ArduPilot/ardupilot #9873: RangeFinder: Using existing  Analog offset for other interfaces (I2C or Serial)

patrickpoirier51, 2018-11-21:
### Feature request
Expand rangefinder offset 

**Is your feature request related to a problem? Please describe.**
Need some easy fix to adjust sensor offset.
For example, the Garmin Lidar Lite, has an offset of approx. 25 cm on low range, and users do not have a mean to compensate.

**Describe the solution you'd like**
Just tested using the existing analog offset  within this code 
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_RangeFinder/AP_RangeFinder_PulsedLightLRF.cpp#L94

    uint16_t _distance_cm = be16toh(val);
    **float offset  = state.offset;**
    // remove momentary spikes
    if (abs(_distance_cm - last_distance_cm) < 100) {
        **state.distance_cm = (_distance_cm-offset);**

**Platform**
[x  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

trilusa, 2018-11-22:
Hi, I think I can do that.
So basically what you are asking for is that the "offset" parameter be accounted for in every sensor driver?
I'm not sure if this is the best approach, but I can just go through each one and subtract "state.offset" from the distance measurement before it is assigned to "state.distance_cm" like you did in your example.
Thanks

patrickpoirier51, 2018-11-22:
Yes that’s it, and I was questioning the method of implementing in each drivers as well.. Could it be processed within rangefinder.cpp or in the header ?