ArduPilot/ardupilot #9885: DSM RC spikes in ChibiOS

Pedals2Paddles, 2018-11-23:
### Bug report
**Issue details**
Random spikes are occuring on all DSM RC channels in ChibiOS.  The timing, duration, and channels all seem to be random.  I found this while trying to figure out why the Solo's gimbal was still twitching even after tridge fixed the serial communication.  Turns out the remaining twitches are in response to these little spikes when they happen to occur on RC6.

Given these these spikes happen on all channels, this could conceivably affect vehicle control, mode changes, failsafes, etc.  They only seem to last a fraction of a second though.

This does not occur in Nuttx master.  Only in ChibiOS.  Which leads me to believe there is still something funky going on in the new decoder.

_These are some screenshots of RC raw input live in QGC._

[image]

[image]

[image]

**Version**
Copter 3.7-dev ChibiOS

**Platform**
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
N/A

**Hardware type**
Cube Black

Pedals2Paddles, 2018-11-23:
@rmackay9 @tridge  I've been watching it on 3.6 this morning and it **does not** experience the DSM spikes.  What is different about DSM in 3.6 vs master?

Pedals2Paddles, 2018-11-23:
I just did a build based on the current 3.6 branch (RC4) and it also does not have this DSM spiking issue.  So I don't think there is an issue in RC to hold anything up.  Whatever is causing it is only in master.

In fact, I cherry picked in Tridge's commits that fix the gimbal's serial comm issue, and the gimbal is now rock solid stable since the DSM issue isn't there too.

rmackay9, 2018-11-23:
@Pedals2Paddles, thank you very much for testing that!  Copter-3.6.2 going out shortly!

tridge, 2018-11-26:
fixed in #9909

closed 2018-11-26 by tridge