ArduPilot/ardupilot #9887: New class of frame Octacpeter hybrid Hexacopter

a2sandres, 2018-11-23:
This frame is similar to the Hexacopter X, only that it has two additional motors continuous to motors 1 and 2 of the arducopter Hexa X and its rotation is contrary to the corresponding motors, this frame is ideal for drones that lift heavy loads.

I think that this frame can be added to the octocopter group, however I do not know what would be the ideal configuration for their behaviors and what values ​​they should give it, for example, the factors in degrees.

We have some copters with this frame configuration, but ardupilot and Mission Planner do not have this configuration, I would like it to be added and I also offer help to do it, however I also look for collaboration for this goal, some flights controller have this configuration and I would be very happy that Ardupilot had it too.

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

Attached sample image

[image]

IamPete1, 2021-02-08:
We now support custom frames via scripting, https://github.com/ArduPilot/ardupilot/pull/15708

closed 2021-02-08 by IamPete1

rmackay9, 2021-02-09:
I guess though that really we should add a script to support this before closing the issue?

IamPete1, 2021-02-09:
Its quite a vehicle dependent mix, not just a variation on motors in a circle like the majority of the mixes, to get it working well it ought to be tailored to each vehicle. I'm happy to take a stab a doing a mixer if @a2sandres is still interested.