ArduPilot/ardupilot #9892: AP_BLHELI: added SERVO_BLH_REMASK possibly breaks BLH support on F405 boards

vierfuffzig, 2018-11-24:
I found servo-type pwm outputs (1100-1500-1900) being reset to motor-type pwm (1000-1000-2000) when restoring parameters from mid october on current fw. could narrow it down to SERVO_BLHELI_MASK parameter. then did a bisection search and could narrow this down to the bug occurring between ArduPilot/ardupilot@c8efff7 and ArduPilot/ardupilot@03f1dec of nov. 15th, that is the adition of reverse motor support.

if MASK is set to non-zero values, multiple SERVOn outputs below MASK value are deliberately set to motor-type and board crashes when setup is fully powered. setting BLH_AUTO = 1 triggers same results.

**Version**
post https://github.com/ArduPilot/ardupilot/commit/c8efff7ddc8073537b20a3f42f80d3325e8ad8bc versions

**Platform**
Plane

**Hardware type**
various F405-type boards (F35, omnibusf4pro, revomini)

**Logs**
https://github.com/vierfuffzig/Chibios-miscellaneous/blob/master/BLH_MASK_ISSUE.tlog

tridge, 2018-11-24:
@vierfuffzig It is deliberate that it changes SERVOn_MIN and SERVOn_MAX to be 1000-2000 for outputs in the SERVO_BLH_MASK. This is needed to ensure that the HAL scales these outputs correctly.
Note that while it sets them, it doesn't save that change, so if you unset the bit in SERVO_BLH_MASK and reboot then the old values will be restored.

> if MASK is set to non-zero values, multiple SERVOn outputs below MASK value are deliberately set to motor-type and board crashes when setup is fully powered. setting BLH_AUTO = 1 triggers same results.

I tested it with SERVO_BLH_MASK=2, which is what you have in the log, and motor two range gets set correctly. Similarly with SERVO_BLH_MASK=4, motor 3 range gets set correctly to 1000 to 2000.
Can you give me an example of where it gets it wrong?

vierfuffzig, 2018-11-24:
thanks for looking into this, tridge. maybe i'm getting things mixed up, but isn't MASK = 2 supposed to set SERVO3 to BLH?
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_BLHeli/AP_BLHeli.cpp#L47
maybe it's only the param description after all...
cheers, basti

vierfuffzig, 2018-11-25:
@tridge i think i understand now that we've switched BLH_MASK and REMASK from the previously documented channel mapping to a binary bitmask. i'd suggest to update bitmask description to avoid confusion: https://github.com/ArduPilot/ardupilot/pull/9898

still imho there's an issue with BLH_AUTO =1 causing board to hang, but that's a different thing.
so i'll rename and close this.

thanks and cheers, basti.

closed 2018-11-26 by vierfuffzig