ArduPilot/ardupilot #9907: Build issues with sitl mabe in a virtual fresh Ubuntu install all the steps could be documented

josheeg, 2018-11-26:
#### Issue details
_Please describe the problem, or desired feature_

#### Version
_What version was the issue encountered with_stable

#### Platform
[  ] Allx
[  ] AntennaTracker
[  ] Copterx
[  ] Planex
[  ] Roverx
[  ] Submarine

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_
Pixhawk cube sitl plane
#### Logs
_Please provide a link to any relevant logs that show the issue_

I have tried building the different ardupilot builds as shown in this step
./waf configure --board px4-v3
https://github.com/ArduPilot/ardupilot/blob/master/BUILD.md

I get 
Setting top to                           : /home/josheeg/ardupilot-master 
Setting out to                           : /home/josheeg/ardupilot-master/build 
Autoconfiguration                        : enabled 
Setting board to                         : px4-v3 
Using toolchain                          : arm-none-eabi 
Checking for 'g++' (C++ compiler)        : /usr/bin/arm-none-eabi-g++ 
Checking for 'gcc' (C compiler)          : /usr/bin/arm-none-eabi-gcc 
Checking for c flags '-MMD'              : yes 
Checking for cxx flags '-MMD'            : yes 
Traceback (most recent call last):
  File "/home/josheeg/ardupilot-master/modules/waf/waflib/Scripting.py", line 159, in waf_entry_point
    run_commands()
  File "/home/josheeg/ardupilot-master/modules/waf/waflib/Scripting.py", line 252, in run_commands
    ctx = run_command(cmd_name)
  File "/home/josheeg/ardupilot-master/modules/waf/waflib/Scripting.py", line 236, in run_command
    ctx.execute()
  File "/home/josheeg/ardupilot-master/modules/waf/waflib/Configure.py", line 159, in execute
    super(ConfigurationContext, self).execute()
  File "/home/josheeg/ardupilot-master/modules/waf/waflib/Context.py", line 204, in execute
    self.recurse([os.path.dirname(g_module.root_path)])
  File "/home/josheeg/ardupilot-master/modules/waf/waflib/Context.py", line 286, in recurse
    user_function(self)
  File "/home/josheeg/ardupilot-master/wscript", line 233, in configure
    cfg.get_board().configure(cfg)
  File "Tools/ardupilotwaf/boards.py", line 751, in configure
    super(px4, self).configure(cfg)
  File "Tools/ardupilotwaf/boards.py", line 44, in configure
    self.configure_env(cfg, env)
  File "Tools/ardupilotwaf/boards.py", line 755, in configure_env
    super(px4, self).configure_env(cfg, env)
  File "Tools/ardupilotwaf/boards.py", line 219, in configure_env
    cfg.srcnode.find_dir('modules/uavcan/libuavcan/include').abspath()
AttributeError: 'NoneType' object has no attribute 'abspath'

Now I figured try the other options see if I get the same result?
Setting top to                           : /home/josheeg/ardupilot-master 
Setting out to                           : /home/josheeg/ardupilot-master/build 
Autoconfiguration                        : enabled 
Setting board to                         : px4-v3 
Using toolchain                          : arm-none-eabi 
Checking for 'g++' (C++ compiler)        : /usr/bin/arm-none-eabi-g++ 
Checking for 'gcc' (C compiler)          : /usr/bin/arm-none-eabi-gcc 
Checking for c flags '-MMD'              : yes 
Checking for cxx flags '-MMD'            : yes 
Traceback (most recent call last):
  File "/home/josheeg/ardupilot-master/modules/waf/waflib/Scripting.py", line 159, in waf_entry_point
    run_commands()
  File "/home/josheeg/ardupilot-master/modules/waf/waflib/Scripting.py", line 252, in run_commands
    ctx = run_command(cmd_name)
  File "/home/josheeg/ardupilot-master/modules/waf/waflib/Scripting.py", line 236, in run_command
    ctx.execute()
  File "/home/josheeg/ardupilot-master/modules/waf/waflib/Configure.py", line 159, in execute
    super(ConfigurationContext, self).execute()
  File "/home/josheeg/ardupilot-master/modules/waf/waflib/Context.py", line 204, in execute
    self.recurse([os.path.dirname(g_module.root_path)])
  File "/home/josheeg/ardupilot-master/modules/waf/waflib/Context.py", line 286, in recurse
    user_function(self)
  File "/home/josheeg/ardupilot-master/wscript", line 233, in configure
    cfg.get_board().configure(cfg)
  File "Tools/ardupilotwaf/boards.py", line 751, in configure
    super(px4, self).configure(cfg)
  File "Tools/ardupilotwaf/boards.py", line 44, in configure
    self.configure_env(cfg, env)
  File "Tools/ardupilotwaf/boards.py", line 755, in configure_env
    super(px4, self).configure_env(cfg, env)
  File "Tools/ardupilotwaf/boards.py", line 219, in configure_env
    cfg.srcnode.find_dir('modules/uavcan/libuavcan/include').abspath()
AttributeError: 'NoneType' object has no attribute 'abspath'

That one seemed to work so I figured why not try sitl at least
well I get this error
osheeg@josheeg-Vostro-220-Series:~/ardupilot-master$ ./waf copter
Waf: Entering directory `/home/josheeg/ardupilot-master/build/sitl'
Command ['/usr/bin/git', 'rev-parse', '--short=8', 'HEAD'] returned 128

What an I doing wrong?

OXINARF, 2018-11-26:
You have to update the waf submodule.

This issues list is for feature requests and confirmed bugs. For development questions please post in the forum (http://discuss.ardupilot.org) or ask in Gitter chat (http://gitter.im/ArduPilot/ardupilot).

closed 2018-11-26 by OXINARF

josheeg, 2018-11-27:
How would I update the waf submodule? I am installing git from the git webpage instructions its not current on ubuntu 18.10

josheeg, 2018-11-27:
sudo apt install git-all

josheeg, 2018-11-27:
how do I update waf? How should I install it?

peterbarker, 2018-11-27:
On Mon, 26 Nov 2018, josheeg wrote:

> how do I update waf? How should I install it?

The fault lies in whatever step you followed before starting on README.md 
- what instructions where you following then?

The submodule should have been correct after the git clone step you did.

You can update the submodules with

git submodule update --recursive

josheeg, 2018-11-27:
what if ubuntu git was a older version I seen that seemed to be a issue in another ardupilot forum thread. 
I got it to compile I put waf in a git message also to try to get it that way. But I was told it comes with ardupilot. I also deleted the ardupilot folder and redid everything in the install sitl in linux tut before readme.md