ArduPilot/ardupilot #9912: BlHeli_32 passthrough initialization of flight controller denied

Naterater, 2018-11-27:
### Bug report

_Please describe the problem_

When the following parameters are loaded on master, BlHeli pass-through does not work. The following parameters, however, DO allow BlHeli Suite to connect on Plane CubeBlack 3.9.3 (stable).

Note reverse thrust is enabled.
Throttle channel (3) for RC input is not centered.
ARMING_REQUIRE=1, otherwise the ESC's do not initialize until arming, and that means they give no voltage telemetry data, etc.

**Version**
Plane latest master

**Platform**
[  ] All
[  ] AntennaTracker
[?] Copter
[x] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Twin-engine ESC's connected to Aux1 and Aux2.

**Hardware type**
Cube black standard carrier board. BlHeli_32 ESC's

**Logs**
Parameters here: https://www.dropbox.com/s/j99uzrhy8mialyp/Master%20Blheli%20Not%20connecting.param?dl=0

IamPete1, 2021-07-26:
I suspect this is now fixed, any update @Naterater ?

Ryanf55, 2025-04-16:
I ran into this today on 4.6 plane. I suspect the ESC I picked up might have had reverse thrust enabled. 

Hardware: Cube Orange +

* ARMING_REQUIRE=1
* BRD_SAFETY_DEFLT=0
* SERVO_BLH_AUTO=1
* SERVO_BLH_MASK=768 (for aux1 and aux2)

I cannot find a workaround yet.

andyp1per, 2025-04-17:
@Ryanf55 can you try https://github.com/ArduPilot/ardupilot/pull/29590 please