ArduPilot/ardupilot #9915: AP_Battmonitor: Add MAV_CMD_POWER_OFF_INITIATED broadcast

Pedals2Paddles, 2018-11-27:
**Issue details**
In the heavily customized version of ArduCopter 3.2ish used on the original Solo cubes, a mavlink command long broadcast of `MAV_CMD_POWER_OFF_INITIATED`  was added in response to the power button on the smart battery.  This is consumed by companion computers and the gimbal.  And it also facilitates powering off the camera.  This function was never ported over to master.  It should be ported over, and usable for any smart battery that can report it's power button state.

The old code can be seen here: https://github.com/OpenSolo/ardupilot-solo/blob/master/libraries/AP_BattMonitor/AP_BattMonitor_SMBus_PX4.cpp#L76

I will work this issue and tag MDB to review.

**Version**
All

**Platform**
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
N/A

**Hardware type**
N/A

**Logs**
N/A

khancyr, 2018-11-27:
It should'nt be broadcasted but forward to the right componant

Pedals2Paddles, 2018-11-27:
In the old code it was broadcast to 0/0.  I plan to use the vehicle's system ID only, but still broadcast to component ID 0.  As such, any component that needs it (now or in the future) will receive it.  To date, I believe the Solo is the only smart battery that reports a power button state anyway.

closed 2019-01-28 by WickedShell