ArduPilot/ardupilot #9922: Update LogAnlyser for Copter-3.6.x

rmackay9, 2018-11-28:
Users are reporting a number of issues with the Auto Analysis tool when used with logs from Copter-3.6.x ([discussion is here](https://discuss.ardupilot.org/t/problems-after-flashing-v3-6-1/35550)):

- The value of LOG_BITMASK is 655358, which is Full + IMU. But log analysis says “No IMU log data”.
- "Found NaN in CTUN DSAlt" even though this is expected when there is no sonar attached
- “THR_MIN not found" because this has been renamed to MOT_SPIN_MIN

peterbarker, 2018-11-28:
On Wed, 28 Nov 2018, Randy Mackay wrote:

>  *  The value of LOG_BITMASK is 655358, which is Full + IMU. But log analysis says “No IMU log data”.

This is a bitmask.

"Full" is 65535

Which includes IMU data.

I don't know what 655358 does.... I've just tested, and it doesn't give 
you IMU or IMU2...

>  *  "Found NaN in CTUN DSAlt" even though this is expected when there is no sonar attached

Pushed a patch in for that.

>  *  “THR_MIN not found" because this has been renamed to MOT_SPIN_MIN

I believe this is already fixed in master.

rmackay9, 2018-11-28:
@peterbarker, great thanks for the fixes.

Re the IMU item, I just copied the LOG_BITMASK number from the user's report.  I haven't verified it but below is a link to a picture of what the user saw.
https://drive.google.com/file/d/1iv7X-TXoygv6vwnLb5AiaZ-Njg5Zhi-C/view

peterbarker, 2018-11-28:
On Wed, 28 Nov 2018, Randy Mackay wrote:

> Re the IMU item, I just copied the LOG_BITMASK number from the user's report. I haven't verified it but below is a link to a picture of what the user saw.

Yep, that value was just incorrect :-)

> https://drive.google.com/file/d/1iv7X-TXoygv6vwnLb5AiaZ-Njg5Zhi-C/view

Just 65535 will do to get IMU and IMU2

closed 2018-12-22 by rmackay9