ArduPilot/ardupilot #9930: Feedback for MAVLink gimbal detection

edwinhayes, 2018-11-30:
### Feature request

In the case of gimbals which have two way communication (primarily MAVLink gimbals I expect, but there are some other serial protocols I think), it would be helpful if there was some user feedback indicating that the flight controller is successfully talking to one.  In the case of aircraft with removable payloads/gimbals, this would be helpful to diagnose issues with payload control.

A single status text message would be fine: "Gimbal detected" or something.  

Something more fancy is possible, but that might require support by ground control software and the sending of additional MAVLink messages, so a single text message seems like it would be fine; it just means that you have access to some information beyond "I waggle the stick and the gimbal doesn't move".

**Platform**
[?] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**

(To be honest, this same sentiment could be applied to a bunch of other areas which would probably be helpful for quickly diagnosing issues.)

IamPete1, 2022-09-11:
@rmackay9 Any thoughts?

rmackay9, 2022-09-11:
AP 4.3 adds a pre-arm check that will display "Mount: not healthy" but the health checks have only been implemented for the Gremsy.  We still need to add them for the Alexmos, SToRM32 (serial and mavlink) and Solo gimbals.

I've recorded this issue here https://github.com/ArduPilot/ardupilot/issues/20985