ArduPilot/ardupilot #9935: Build: Add new Solo targets to auto build system

Pedals2Paddles, 2018-12-02:
### Feature request
Two new build targets for the Solo (`CubeBlack-solo` and `CubeGreen-solo`) will be added when https://github.com/ArduPilot/ardupilot/pull/9932 is merged.  I ask the dev team to consider adding these two building targets to the auto build system.  This will give all Solo users easy access to the proper firmware directly from http://firmware.ardupilot.org.  It makes the proper firmware with the proper parameters easily accessible for manual loading or for GCS apps to download at all times.  This will be effective for Copter 3.7 onward.  There is no expectation to backport to 3.6, although I wouldn't mind it happening either.

I fully appreciate not wanting to open the floodgates for everyone that wants to have their own version pre-made for them.  And I definitely do not want that to happen.  I believe the Solo is worthy of this because it is totally unlike most (if not all) other vehicles running ArduPilot.  No other user or vehicle that I'm aware of with ArduPilot can check all these boxes.  So in my opinion, this really mitigates the "_you did it for Solo so why not for me_" issue.
- Thousands of potential users of this new auto build, not just a one-off.
- Mass produced, commercially available, consumer RTF product
- Loading firmware that doesn't have the proper parameter defaults baked in can brick the vehicle.
- Loading firmware that is not of the correct type will brick the vehicle.
- Resetting parameters or factory resetting the Solo can brick the vehicle without proper defaults.
- **Troubleshooting or fixing firmware loads requires complete teardown, which most users cannot do.**
- **Loading parameter files, messing with Mission Planner, etc is not within the skillset of most users**

OXINARF, 2018-12-04:
> Thousands of potential users of this new auto build, not just a one-off.

How many Cube Green autopilots have been sold? That's a good metric to know the potential target for this. Thousands of potential users is quite vague.

> Mass produced, commercially available, consumer RTF product

Solo isn't commercially available anymore (at least widely), as the Iris isn't anymore. There are others though, as the Skyvipers and the Foxtech Nimbus. I don't think this is important though, it's more important to know how many users will actually use it.

> Loading firmware that doesn't have the proper parameter defaults baked in can brick the vehicle.
> Loading firmware that is not of the correct type will brick the vehicle.
> Resetting parameters or factory resetting the Solo can brick the vehicle without proper defaults.

Break the vehicle in which way?

> Troubleshooting or fixing firmware loads requires complete teardown, which most users cannot do.

Since when? Solo has a CC which can upload the firmware...

> Loading parameter files, messing with Mission Planner, etc is not within the skillset of most users

Then ask the GCS author to simplify it... Also, that would be a generic problem, not a Solo one.

Pedals2Paddles, 2018-12-04:
Elaborating on some discussion from the dev call.... **I do need the entire build to be run**, since it is no different than the CubeBlack build other than parameters.  If the auto build system can simply copy the CubeBlack APJ file, apply the parameter defaults, and spit it out with a new name, that's really all we need.

khancyr, 2018-12-04:
I would prefer the second option for two reason : 1 it will be simplier and don't waste build time, 2 it will allow us to have a good example on how to use the apjtool!

Naterater, 2018-12-05:
Furthermore on the dev call there was discussion on what to call these builds. Considering that there are "tens of thousands" of original (green) autopilots out there with users who probably know nothing about the fact that they can change their flight controller, I think the firmware names should be something like the following:

1. Solo
2. Solo Cube Black Upgrade

Don't call the first build "green" because it will only confuse the majority of the thousands of people who don't know much about their systems and still have everything original. How many people even know that it's green inside?! Anybody who has upgraded to a Cube Black will know to use the other firmware.

And yeah, I guess use the new .apj system to just change the parameters for each build. That sounds easy?

My $.02

Pedals2Paddles, 2018-12-05:
It's the other way around. Green is the custom install with 5v signalling people would have bought and installed. Black is the standard cube, whether it be the original 3DR built plastic one or the newer Hex made metal one.  They're both black in color and use the same firmware and parameters.

The naming convention on firmware.ardupilot.org would dictate the files be called `CubeBlack-solo.apj` and `CubeGreen-solo.apj`.  I'm not opposed to a more user-friendly name, such as `Solo.apj` and `SoloGreen.apj`.  However, users interacting with the files directly would be following instructions that tell them which file to download anyway, so it may be best to stick with the naming conventions already in use.   Sticking with the existing naming standards would also cause heartburn for the people who are vehemently opposed to anything related to this by default...

I also absolutely support not duplicating the build.  Just having the auto-build scripts copy the existing CubeBlack apj file, change the default params, and upload it to the firmware server with the appropriate file names would be perfect.  That operation would (I think) have a de minimis effect on the auto-build scripts and firmware servers.

Pedals2Paddles, 2018-12-05:
Another great reason to have the auto build firmware available on the firmware server, which I did not articulate in the issue description, is **testing availability**.  Anyone with a Solo can grab the latest master and load it at anytime to test the new features and patches.  It will just work properly like a consumer product should, and will not brick the vehicle.  There are lots Solo users who would love to be test flying master all the time, and ArduPilot would be gaining all those test pilots.

As it stands today, that doesn't happen unless I take a master build, load the parameter defaults, and push it out through an application for users to test once in a while. That is extremely limiting to the users and extremely limiting to ArduPilot's testing base.  It is also not an efficient use of hours of my time, which the auto-build server can do in a few seconds.  And like Tridge said in the other PR, it eliminates countless points of human error.

This is a substantial benefit to ArduPilot.

proficnc, 2018-12-05:
The naming convention is well understood by the community.
The cube Green is the solo cube.

Let’s please not confuse the conversation with deviation from the topic.

There is already a cube Black build, a cube orange build, and we have a special cube purple build coming 

The cube green is simply an extension from that.

The ability to bake the parameters in reduces the load here on the whole community. 

This is a very good thing that Matt is doing

Pedals2Paddles, 2018-12-11:
Thank you everyone that's helped with this.  Per the dev call this evening, closed with https://github.com/ArduPilot/ardupilot/pull/9932 and https://github.com/ArduPilot/ardupilot/pull/9987.

closed 2018-12-11 by Pedals2Paddles