ArduPilot/ardupilot #9939: SBUS Out not working

GlidLov, 2018-12-03:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

### Bug report

SBUS Out port (next to "cons" port) is not working on any 3.6.x (i think it's an old issue, but can't test on older versions) on the Pixhawk 2.1 "cube".
The only way to get an SBus out is to use the serial ports, assigning one to work as Sbus out.
I tested with an oscilloscope and SBUS out port is never working, all channels are "0". I tried all the possible parameters (refresh rate, etc), no changes.

**Version**

3.6.x

**Platform**
[ X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Hardware type**

Pixhawk 2.1

GlidLov, 2018-12-05:
update: 

it's not working if BRD_SAFETY_ENABLE = 0

matthewgan, 2019-04-24:
Same problem.
It's not working in BRD_SAFETY_ENABLE = 1 and BRD_SBUS_OUT = 1 here.
on Pixhawk 2.1
FIXED WING

matthewgan, 2019-04-24:
Found reason in the code:
in order to enable the sbus output, you must arm it to pass the if (status.io_enable())
so it is somehow not a bug, but designed that way.
Who designed it please explain why doing this?

rmackay9, 2019-04-25:
@matthewgan, ok great that you got to the bottom of this.

I think the underlying issue is perhaps the defaulting of the BRD_SAFETY_MASK is not ideal on Plane.  This parameter controls which outputs are affected by the safety switch.  The screen shot below of the MP shows that if the user clicks on the parameter's Value from the Full Parameter Tree screen the individual outputs are affected by the safety switch (checked channels will move regardless of the state of the safety switch).

[image]

rmackay9, 2019-04-25:
I've modified the subject line of this issue a bit, hope that's OK.

IamPete1, 2022-09-11:
I'm going to close, the users original issue has been resolved and I don't think we would ever be brave enough to change the default safety behavior.

closed 2022-09-11 by IamPete1