ArduPilot/ardupilot #9946: Plane 3.9.3: uavcan airspeed doesn't work

liang-tang, 2018-12-04:
### Bug report
**uavcan doesn't work on plane 3.9.3, but works on master branch.**
flashed plane 3.9.3 version, enabled uavcan and set airspeed type to 8, reboot.  no airspeed on hud.
[image]

just flashed master branch version, didn't change any param, connected to mission planner, airspeed works.
[image]

**Version**
plane 3.9.3, Chibios version

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[*] Plane
[  ] Rover
[  ] Submarine

**Hardware type**
The cube.

EShamaev, 2018-12-04:
AP_Airspeed_UAVCAN is not in 3.9.3

[image]

liang-tang, 2018-12-04:
OMG，I am sorry. I did not notice.

closed 2018-12-04 by liang-tang