ArduPilot/ardupilot #9964: Rover: 3.5-dev initial mode parameter is not working

rmackay9, 2018-12-07:
seems like Rover's INITIAL_MODE parameter is being overwritten by the first RC input from the transmitter.

tajisoft, 2018-12-20:
@rmackay9 Hello. In my understanding, if RC is connected, I think it should be change mode with mode switch position. ArduPlane also has INITIAL_MODE parameter. I think same phenomenon is occurring. Could you give me some advices for me. I want to fix this issue.

rmackay9, 2019-09-18:
After discussing some more with @peterbarker and @Hwurzburg we think the transmitter should always win so INITIAL_MODE should only apply when the RC is not being used.

Hwurzburg, 2019-09-18:
so auto-triggered RC-less mode can't work....as seen here...https://discuss.ardupilot.org/t/automatic-start-with-auto-trigger-pin-how/27146...I expect upon initialization (and maybe INITIAL_MODE is actually set then), but lacking RC it goes into failsafe...my guess is that unless FS_ACTION is set to DO NOTHING, any RTL mode change will fail due to EKF/GPS and it tumbles into HOLD...the INITIAL_MODE has to be set after FS mode change occurs, for the first time after init, but never again..you want to have it set for no RC or GCS cases...
also, I dont understand how armed/disarmed plays into all of this in the auto-triggered cases (switch or kick) unless those are themselves arming cases in the code and also the RC/GCS pre-arm checks are disabled...lots of questions..

davidbuzz, 2019-10-01:
I'm having a similar issue with INITIAL_MODE not being honoured... but ive at least got a patch that fixes my specific use case...
My use case is basically @Hwurzburg 's option 1... where i have cube hardware that will quite litterally have power applied, and with zero user interaction, it needs to enter AUTO at the right time ( which is clearly after it's got gps and the ekf reports stable.    I have a patches for my use case, ie when INITIAL_MODE=10, it works, and puts the vehicle into AUTO after the EKF reports that it's locked onto the gps, and so we know the gps has reached thresholds of sanity as well. https://github.com/davidbuzz/ardupilot/commit/19007b46ff2107ad04d3c1b59505a12a362ef61d