ArduPilot/ardupilot #9969: PILOT_TKOFF_DZ parameter dont work

AlvaroRAS, 2018-12-07:
**Issue details**
PILOT_TKOFF_DZ, dont work, I change it to 500, but the copter still trigger takeoff at 100.

**Version**
Copter QUAD 3.6.3

**Platform**
[  ] All
[  ] AntennaTracker
[X ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad

**Hardware type**
Pixhawk 2.1 CUBE

peterbarker, 2018-12-31:
This parameter seemed to be completely unused.

The current takeoff code looks for a desire to climb, not a throttle input value.

I've added a patch to one of my PRs that makes us look at this parameter.

I'm actually unconvinced - if nobody has missed the parameter, perhaps we should remove it (@jschall would know why it was added...)

rmackay9, 2018-12-31:
I'm OK with us just removing it.  One person has noticed that it doesn't work.. but we went through a lot of testing with nobody noticing.

peterbarker, 2018-12-31:
On Fri, 7 Dec 2018, AlvaroRAS wrote:

> PILOT_TKOFF_DZ, dont work, I change it to 500, but the copter still trigger takeoff at 100.

What was your use case for this parameter?  As you can see, we're thinking 
about killing it completely :-)

AlvaroRAS, 2019-01-01:
Hello guys and Happy New Year !!!, the reason why this parameter could be important for us: is that our company modify our radio control (Horus 12) to have the auto-centered throtle stick function to made our copters more dji like (because our clients are very familiar with dji products, and we are looking to made our copters easy to use for them).

We use also the parameter PILOT_TKOFF_ALT = 300 and PILOT_THR_BHV = 4, with these combination of parameters the copter takeoff to 3m from the ground just by rising the throtle stick a few up from the midle, and the landing proces is very easy also and safe by using the parameter PILOT_SPEED_DN = 150.

Everything was fine with some clients, but sudendly apears one client that after arming the copter he release the trotle stick fast and the auto-center configuration at the radio-control, put the stick immediately above the PILOT_TKOFF_DZ parameter, in concequence, the copter started its take off procedure inmediatly and the copter almost flip at the ground, because the motor didn't have enought time to start spining (14kg copter, low KV motor, 27" propellers);

So the  PILOT_TKOFF_DZ at 300 could prevent this escenario to happend again; our company also figure out other parameters and characteristics that copter firmware could have, of couse I will add them in "Pull Request" zone (configurable delay parameter before take off procedure (for big low KV motors to start), configurable delay parameter for the copter to be armed (for let all the sensors and external sensors like magnetometers for magnetometer survey at the copter be ready), other one is a parameter that can handle the cicling start of the motors one by one this could be very usefull to reduce current spikes at take off for big copters 40kg+.

Thank for your attention, and have a Happy New Year !!!

rmackay9, 2019-01-01:
We have a parameter called MOT_SPOOL_TIME that may help.  This controls the time it takes for the motors to spin up.

rishabsingh3003, 2020-03-05:
This parameter was completely removed in : https://github.com/ArduPilot/ardupilot/pull/11039 .

rmackay9, 2020-03-05:
I was going to close this but the enhancement request is still valid.  We went ahead and removed the parameter after the feature was removed.. but that doesn't make the request any less valid.

.. so I think I'll reopen this.

rishabsingh3003, 2020-03-05:
My initial thought was that we could close this issue since the parameter mentioned in the description has already been removed, which makes it unclear for the new users who land up here on what the issue really is. Maybe we can open up a new issue which explains the enhancement request.

rmackay9, 2020-03-05:
@rishabsingh3003, yes good point.  I've modified the subject line of this enhancement request to hopefully make it more clear