ArduPilot/ardupilot #9970: Ardupilot x Storm 32 and the tilt angles

JeyPi, 2018-12-07:
### Bug report

**Issue details**
Storm 32 and the tilt angles 

The bug is I think in the way the  arduplane software computes the tilt angles. 
When in Auto on a mission I must design the track within certain limits (height and distance) from the region at which the camera should be aimed.
Example:
If I am at a 100 meters above and 60 meters away, the mavlink sends a correct tilt angle for a good aim at a given location.
But if I descend to 10 meters above being still 60 meters away, the camera is pointed too low. Or if I keep the 100 meters and get away lets say 300 meters, the camera is pointed too high. 
This must be an easy mathematics in the code that has been working improperly.
But nothing is nicer then when in auto flight the planes camera points exactly at given positions during the planned flight. I can still do it but as I said I need to plan the track well to keep the distance - height relation to ROI within the limits where the arduplane miscounting does not take such an effect.

I also am curies about where to find the camera tilt yaw roll numbers int he log evaluation 

Another of my experience:
Not many people are using mavling during an Arduplane Auto mission where there are ROIs determined for the Storm 32 controller aiming the SBG.
I am using mavlink to have the Pixracer communicate with the Storm 32.
If I set these parameters like this:
MNT_ANGMIN_PAN = -18000, MNT_ANGMAX_PAN = 17999
MNT_ANGMIN_ROL = -6000, MNT_ANGMAX_ROL = +6000
MNT_ANG_MIN_TIL = -9000, MNT_ANG_MAX_TIL = +9000

My camera points too low.

I have to adjust the parameters like this to have the camera point with the right pitch:

Storm 32 Pitch settings:
‘Rc Pitch Min’ => 120.0 °
‘Rc Pitch Max’ => -90.0 °
‘Rc Pitch Offset’ => -49.0 °

MNT_ANGMIN_PAN = -18000, MNT_ANGMAX_PAN = 17999
MNT_ANGMIN_ROL = -6000, MNT_ANGMAX_ROL = +6000
MNT_ANG_MIN_TIL = -11000, MNT_ANG_MAX_TIL = +11000

I always set the ROI altitude relative to the height of the flight. If the plane is 100 m above the ground circling the ROI, the ROI altitude is set to 0.
To have the camera actually point with the right pitch, I need to adjust the Storm 32 pitch offset in the Storm 32 gui. That way it uses a good pitch angle if I am 100 meters above and away from the object, but the further the plane is, the more is the camera looking up in the sky.
Is this supposed to be a bug that the arduplane is sending wrong mavlink data to the Storm 32?

Version
Arduplane 3.8.4
Storm 32 v0.96
Platform
Plane
Airframe type
Motorised glider
Hardware type
Pixracer,
Logs
Will add one if needed. This situation is easily reproduceable

IamPete1, 2022-09-11:
@rmackay9 another gimbal issue, I wonder if we can close now?

rmackay9, 2022-09-11:
@IamPete1, @JeyPi,

I've never heard of this issue before (sorry I missed it).

The ROI altitude should be set to the altitude above home (not relative to the vehicle).

There were a number of issues with the SToRM32 mavlink interface in ver0.96 (of the gimbal) so I think it would be good to upgrade the gimbal to a more recent version of the SToRM32 software and re-test with ArduPilot 4.3 (which will start beta testing very soon).  If the gimbal can't be upgraded then set MNT1_TYPE = 5 (SToRM32 serial).

By the way, [the wiki](https://ardupilot.org/copter/docs/common-storm32-gimbal.html) has been slightly updated to clarify the SToRM32 and AP version dependency.

rmackay9, 2022-09-11:
I think I will close this but we can re-open if the above advice doesn't help.  I'll see replies on this issue.

closed 2022-09-11 by rmackay9