ArduPilot/ardupilot #9972: AP_Compass Does Not Probe For MPU-9250 When HAL_COMPASS_DEFAULT Is Set To HAL_COMPASS_NONE

davidbitton, 2018-12-08:
### Bug report

**Issue details**

speedybeef4 fails initial boot with 
> No Compass backends available
> COMPASS INIT ERROR

This is related to issue #5496. I am using a combination GPS/IMU/Compass; an [RY836AI](https://tro.pe/2SFDmNl) from [Reyax](https://tro.pe/2Sxv21Z). The compass is an MPU-9250. I've made headway with the code behind this and I've added that to the original issue.

**Version**

github master

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**

Quad

**Hardware type**

speedybeef4

**Logs**

-none-

IamPete1, 2021-08-29:
@davidbitton any update?

davidbitton, 2021-08-30:
> @davidbitton any update?

from three years ago? I'll grab the speedy i have and load master.