ArduPilot/ardupilot #9981: AP_Proximity: RPLidarA2 waits for undocumented message to start (not compatible with A1 model)

Brooce, 2018-12-10:
### Feature request

**Is your feature request related to a problem? Please describe.**
I've been trying to run the RPLidar A1M1 as a proximity sensor. It's an outdated version, but the application note with communication protocol is the same for A1 and A2, so this shouldn't be a problem. However, AP_Proximity waits to receive 62 bytes of sensor information after restart before proceeding to start the scanning. It turns out that my device is sending only 56 bytes!

I couldn't find any information in official documents regarding this "welcome message", I suppose the bytes number was found experimentally. Thus, even though this issue concerns an outdated lidar model, such approach may lead to a serious bug if the message changes in another firmware release for A2. The same applies to the first character of the expected message (arbitrarily set to 0x52). Fixing it might also be an easy way to expand library's usability for popular A1 model.

**Describe the solution you'd like**
System's performance shouldn't be based on a non-official information. Vendor's application note suggests to simply wait 2ms after sending the RESET command:
> Since RPLIDAR won’t send response packet for this request, host systems should wait for at least 2 milliseconds (ms) before sending another request.

Nevertheless, the full welcome message takes about 5ms to broadcast.

**Describe alternatives you've considered**
Finding a more flexible index of receiving full welcome message could do the trick, but it's still basing on  undocumented behavior.

Changing the expected bytes count to 55 makes the code work with RPLidar A1M1.

**Platform**
[ x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
A faulty line can be found in **AP_Proximity_RPLidarA2.cpp** file at **AP_Proximity_RPLidarA2::get_readings()** function:
`if ((c == 0x52 || _information_data) && _byte_count < 62) {`
This makes the state machine stay in _rp_resetted_ state until 62 bytes are received (what never occurred in my case). 

The "welcome message" that my lidar sends is (with CR+LF as newline characters):
>RP LIDAR System.
Firmware Ver 1.15, HW Ver 0
Mode: 0

I'm curious what is the 62-bytes message like? The only information I managed to find in google is this blog post with the very same message: http://andrewkoroluk.com/blog/57f4933c0f9923a86cf67a19

rmackay9, 2018-12-10:
I wonder if Steven Josefs has any idea about this.  I don't see his github username so perhaps he won't see this..

Brooce, 2018-12-10:
Thanks for paying attention to my request, Randy. Do you think the issue is significant enough to be worth notifying him some other way?

amilcarlucas, 2018-12-10:
Hi there, I will make sure Steven gets the message.
But the easiest way to get this moving is for you to open a PR.

Brooce, 2018-12-12:
Thanks! Yeah, I've already fixed it: after the RESET request, Copter waits 10ms for Lidar to stop transmitting (2ms as per datasheet, but we need at least one loop execution to flush UART buffer). Then it starts to listen for the expected _welcome message_, and waits 10ms since the first character for it to finish.

Will make a PR as soon as I'm certain that everything works. But it will need to be proved with someone using A2 lidar to make sure the timing is fine. I could use waiting for arbitrary characters to make the code more bulletproof, but that's just what I wanted to omit for compatibility issues...

Brooce, 2019-01-22:
Here's a PR with some more updates: #10303

peterbarker, 2024-07-31:
We fixed A1 years agao - time to close this :-)

closed 2024-07-31 by peterbarker