ArduPilot/ardupilot #9983: Reverse Thrust ESC RPM reporting positive

Naterater, 2018-12-10:
### Bug report

RPM is reported as positive even during reverse thrust is active. Although I believe this may be a limitation of telemetry ESC's, I think ardupilot should switch the sign and log RPM as negative during reverse throttle application to the ESC.

**Version**

Plane 3.10-latest  roughly 2018-12-02

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Believer

**Hardware type**
Lumineer 51A BlHeli_32 Telemetry ESC's

**Logs**
https://www.dropbox.com/s/x3lk1ezq49lagsr/00000018.BIN?dl=0

auturgy, 2018-12-10:
This might make particular sense for rover, where rpm could be used in lieu of a wheel encoder.

Regards,

James

Naterater, 2018-12-11:
Hey @rmackay9 wouldn't this actually be for all vehicles supporting BL_Heli - This could eventually be used for copters flying upside-down and like @auturgy mentioned, rovers too. If you agree, I'll edit my above post to reference all of these platforms.

IamPete1, 2021-08-29:
This would require a MAVLink change, the current esc RPM message uses `uint16_t` We could add a direction bitmask as a extension and leave it to the GCS to display correctly I guess.

auturgy, 2021-08-29:
FYI the esc_status message in upstream common allows for negative values.