ArduPilot/ardupilot #9984: .BIN log dates to 1980

Naterater, 2018-12-10:
### Bug report
.BIN files are being saved with a 1980 date, and it appears to be random. When using the log analysis tools and geotagging, the 1980 date sticks. The GPS link on these flights was solid.

**Version**
Plane 3.9.3 ChibiOS
Plane 3.10 - Latest (late November)

**Platform**
[?] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Believer

**Hardware type**
Cube Black Standard Carrier Board. Here GPS and Here+ GPS (separate builds).

**Logs**
_Please provide a link to any relevant logs that show the issue_

rmackay9, 2018-12-11:
MP issue I think so I'll close if that's OK.

Naterater, 2018-12-11:
No, this is the actual binary logs... 
[image]

[image]

rmackay9, 2018-12-11:
@Naterater, but these are logs downloaded using MP right?   Have a peek on the SD card directly and you'll likely see that they don't have a date on them, they are all just numbered.  It's MP that changes the name of the file as part of the download.  I won't close this issue again 'cuz we don't need to get into a tug-of-war.. but I'm pretty sure it's an MP issue.

Naterater, 2018-12-11:
These are straight from the SD card - NO MP intervention at all.

Naterater, 2019-12-10:
Haven't seen this issue lately in the 4.0 releases.

closed 2019-12-10 by Naterater