ArduPilot/ardupilot #9993: Strange new issue with landing in Plane master build.

pompecukor, 2018-12-11:
### Bug report

**Issue details**

Something fishy going on in the latest arduplane master. Very easy to reproduce in Mission Planner SITL, however I first noticed it in real flight with 2 separate units that are running the latest master.

In short the problem is that on any mission with auto landing. Once the plane passes the last waypoing, before the land point. It rolls right and does a semi loiter, then rolls back left and heads for the landing point and lands.

This is very easy to reproduce, all you need is any random mission, I am including one for your convince set at a random location. As you can see WP4 is a waypoint, then WP5 is tha land point. Once the plane passes WP4 it goes right and then left. 
Just run on Mission planner simulator. Select Fixed wind plane and run the mission. You will get the same result. I have tried it 5 times with different mission. As mention it happens in real life too. So not MP issue.

NOTE: Another strange thing is that it will only happen the first the first time you run it. If you restart same mission with the same plane it will not happen the second time. You need to disconnect and start simulator again for it to happen again.

**Version**
Latest Arduplane master build.

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Any fixed wing, including simulator

**Hardware type**
Irrelevant. happend on Sim too.

**Logs**
As it can be reproduced in sim anyone, no need for dataflash logs.

[image]

[master land issue.zip](https://github.com/ArduPilot/ardupilot/files/2668195/master.land.issue.zip)

[image]

pompecukor, 2018-12-11:
In this picture you can see what happens if you run the exact same mission, without disconnecting the plane. That is in Simulator. right after it lands. Just press restart mission and arm (if needed).
This time it will fly as expected. No hard right or any other turns when going from WP4 to WP5 (land).

[image]

vierfuffzig, 2018-12-11:
yup, noticed that too the other day, but thought i messed something up...
for some reason on 1st flight of the same mission there's a bearing demand introduced that isn't present anymore when flying that same mission a second time:
[image]

don't know if this is related, but i noticed NAV_LAND's param4 field (desired yaw angle) is forced to 1 during waypoint handling, which is not exactly self-explanatory on fixed wing:
[image]

https://github.com/vierfuffzig/Chibios-miscellaneous/blob/master/SITL_Plane_2landings.BIN

tridge, 2018-12-11:
thanks, I've reproduced it in SITL
I've also added @WickedShell to the issue, as I suspect it is related to some new landing features he has been working on

magicrub, 2018-12-11:
yeah, I suspect it's @WickedShell 's recent change too. I'll ping him

WickedShell, 2018-12-12:
This is not from the recent quadplane landing stuff I've been doing. I had to use @pompecukor's mission to get it to break. Git bisect shows it broke in: 966b359999590a358 (AP_Landing: yaw correction on landing)

tridge, 2018-12-12:
I've reverted the AP_Landing change that causes this issue
ping @EShamaev

closed 2018-12-12 by tridge

EShamaev, 2018-12-20:
Looking into this, the problem might be outside of the landing code, but was revealed by the changes.