ArduPilot/ardupilot #10001: Pixhack V3 ST LED and acceleometer calibration

xabierblanch, 2018-12-12:
### Feature request

**Is your feature request related to a problem? Please describe.**
The pixhack V3 ST LED is fixed red after the boot and the accelerometer can not be calibrated. An error appears in MP. 

**Describe the solution you'd like**
Can it be a problem with new versions of Arducopter? With previous versions (I updated recently) it did not happen.

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

xabierblanch, 2018-12-12:
With px4 firmware works perfectly. But I'm interested in carry out ArduPilot. (for me is better)

rmackay9, 2018-12-13:
Could you raise this in the Copter support forums first?  https://discuss.ardupilot.org/c/arducopter/copter-36.

closed 2018-12-13 by rmackay9