ArduPilot/ardupilot #10006: Mission Planner displays Bad Logging but LOG_BACKEND_TYPE=0

otsuka-kohei, 2018-12-13:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

### Bug report

**Issue details**

When I arm PX4 with healthy FAT formatted microSD card but Mission Planner displays "PreArm : Logging Failed" and "Bad Logging".
Therefore I set LOG_BACKEND_TYPE to 0 (disable logging).
But Mission Planner keeps displaying "Bad Logging".

**Version**
ArduCopter V3.6.3 Quad

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
quad X

**Hardware type**
Pixhawk PX4 2.4.8

**Logs**

dposea, 2018-12-13:
I was going to open an issue on this also. It frsky passthru checks for an open log before arming, which is not good. If logging fails an error message is already generated. Do we need this at all? Passthru works while generating errors every few seconds. I’ll fix it if  removing it is ok.

peterbarker, 2018-12-13:
On Wed, 12 Dec 2018, Otsuka Kohei wrote:

> When I arm PX4 with healthy FAT formatted microSD card but Mission Planner
> displays "PreArm : Logging Failed" and "Bad Logging".
> Therefore I set LOG_BACKEND_TYPE to 0 (disable logging).
> But Mission Planner keeps displaying "Bad Logging".

Did you reboot after making tht setting?

Are you sure it's "Bad Logging", not "Logging Failed" MP is showing after 
a reboot?  If it is the latter then you need to tell the vehicle it is OK 
to arm if it is not logging by modifying LOG_BITMASK

peterbarker, 2018-12-13:
... where by `LOG_BITMASK` I mean `ARMING_CHECK`, obviously.

peterbarker, 2018-12-13:
On Thu, 13 Dec 2018, David Posea wrote:

> I was going to open an issue on this also. It frsky passthru checks for an
> open log before arming, which is not good. If logging fails an error message
> is already generated. Do we need this at all? Passthru works while
> generating errors every few seconds. I’ll fix it if removing it is ok.

Is this actually the same issue as the OP?  He didn't mention FrSky at 
all.

otsuka-kohei, 2018-12-14:
I rebooted PX4 after update parameters. 
Mission Planner sometimes shows "PreArm : Logging Failed" before "Bad Logging".
And I set LOG_BITMASK to 0 (disabled) but Mission Planner shows these error messages.

I am using SanDisk 4GB speed class 10 (not UHS) microSD, so I will buy UHS1 or faster microSD.

peterbarker, 2018-12-14:
On Thu, 13 Dec 2018, Otsuka Kohei wrote:

> I rebooted PX4 after update parameters.
> Mission Planner sometimes shows "PreArm : Logging Failed" before "Bad Logging".
> And I set LOG_BITMASK to 0 (disabled) but Mission Planner shows these error messages.

Setting LOG_BITMASK to 0 doesn't really disable anything; we still log 
things even if it is zero.

Are you saying that if you set LOG_BACKEND_TYPE to 0 and reboot you're 
getting "Logging failed" messages?

otsuka-kohei, 2018-12-14:
Mission Planner shows "PreArm : Logging Failed" with LOG_BACKEND_TYPE=0.

peterbarker, 2018-12-14:
On Thu, 13 Dec 2018, Otsuka Kohei wrote:

> Mission Planner shows "PreArm : Logging Failed" with LOG_BACKEND_TYPE=0.

OK, that is expected unless you remove logging from the ARMING_CHECK 
bitmask.  Have you done that?

If you haven't done that you're effectively telling the vehicle, "you must 
be logging to arm - but I'm not going to tell you how to log"

rmackay9, 2018-12-14:
@otsuka-kohei, Sorry for being annoying but a little semantic correction.  "Pixhawk" is hardware, "ArduPilot" is software.  So we reboot the "Pixhawk" after changing a parameter.

dposea, 2018-12-14:
"Bad logging" only comes from frsky passthru . I ran a find with grep to check all files. I always get "Bad Logging" until I arm. It's checking sensor_status for MAV_SYS_STATUS_LOGGING, and I haven't tracked that down fully yet. Until I checked the source code I thought it was a ChibiOS problem. It was not happening with NuttX. I can open a new issue when I have all the details since this seems to be a different issue.

peterbarker, 2020-11-25:
I believe this was addressed.

closed 2020-11-25 by peterbarker