ArduPilot/ardupilot #10036: Full servo control, inputs, outputs, full house wing, complex setup.

SlipstreamFPV, 2018-12-17:
### Feature request

Full servo control now that servo output is separate from RC Inputs.

**Describe the solution you'd like**

Making a full house glider or other complex setup work with Arduplane is almost impossible.

Now that we have separate servo output control it should be easy, however servo center points, servo reverse and servo endpoints are still not respected depending on which servo function you use. For example if flaps (Function 2) are set as the function and a mid point is set on the servo out function then this should be respected as the middle of the flap range when the flap input RC channel reaches the set trim value. This way if you're setting up to indepedant flap channels for each side of the wing you can set midpoints and endpoints that align.

Further to this, a slave input function would be another great addition. Some wing functions require aileron and flap input or reverse flap input. The falperon mixer is okay, however when you want to use two sets of flaperons or other complex setups like full wing camber it just doesn't cut it because you can't do multiple different setups in different directions. Each servo output should have it's own option for an aditional flap input channel.

**Describe alternatives you've considered**

Tried most things. Tell me if it works and I'm wrong!

**Platform**

[X] Plane

IamPete1, 2018-12-17:
https://github.com/ArduPilot/ardupilot/pull/9391

Differential spoilers and crow breaking recently made it into master, not sure if it will help in this case or not.

Naterater, 2018-12-17:
I think this should be broken down into separate issues. The current logic in master that will soon make its way out into stable releases for at least some of the mixing that you are asking about adding. I'm going to summarize the issues that you have brought up, but in my own words...

1. SERVOn_TRIM and RCn_TRIM has no effect on flap mid-point. It only respects SERVO and RC MIN and MAX values. Is this correct?

2. SERVOn_FUNCTION values 24 and 25 (Flaperon Left and Flaperon Right) do not function properly. I don't have experience with this, and I haven't heard any other complaints about it. Maybe you could provide some more information about exactly what you are trying to do so that we could help.

It's difficult to provide complex features like custom mixing that are still simple enough for novice users to understand. Parameters can be difficult to set up, and with hundreds of them in the list, it can be overwhelming fast.

SlipstreamFPV, 2018-12-17:
In addition to differential spoilers and crow braking in the latest master do flaps/auto_flaps continue to work like normal on the inside wing control surfaces?

IamPete1, 2018-12-17:
There is no documentation for this yet but it looks at though flaps work as before and you get four new dspoiler outputs, that I guess are a combination of rudder ailerons and flaps. [This ](https://discuss.ardupilot.org/t/crow-braking-on-flying-wings-now-working/35407)is the place to find out more I guess.  I suggest you flash latest and try setting them up, they may do most of what you want. Then we can look at any changes from there.

Naterater, 2018-12-17:
I think there are 2 parts to this discussion, and the first about mixing belongs here: https://github.com/ArduPilot/ardupilot/issues/6422

The second issue - flaps not having a trim value - could be an issue, but I don't think it's that critical. If you want to change this conversation to discuss that, we can change the title, but ultimately I recommend opening focused issues with specific issue titles (Plane: SERVOn_TRIM has no effect on flap mid-point). Make your case about wanting camber.

Why don't you operate with flaps at 25% at all times when using flaperons? That way you can have reflex when you go to 0% flaps and plenty of traditional flaps at 100% flaps? Anything more is very uncommon and not applicable to most aircraft or users.

Ryanf55, 2019-01-12:
Hello Naterator,

I'd like to add to the conversation from #8009 . The plane I was working on exceeds the assumptions you make; most properly designed full house wings will. For example, the flaps were centered around 2100uS and 950uS, which is below 25% or greater than 75% travel. This gives the flap servo maximum resolution and travel. I personally haven't seen people use more than a few % of reflex on the flap surfaces, so that is why the servo is so offset. No matter the way development goes, I'd like to have any hard coded limits or assumptions for the new mixer. That seems to defeat the purpose.

Naterater, 2019-01-17:
Is it alright if I close this issue in favor of #6422 since this is about mixing in general? I think this conversation belongs over there, and I appreciate the comments you have left there @Ryanf55.

IamPete1, 2019-04-03:
https://github.com/ArduPilot/ardupilot/pull/10614 has been merged so this is now possible

closed 2019-04-03 by IamPete1