ArduPilot/ardupilot #10039: RUDD_DT_GAIN parameter has no effect on quadplanes with tiltable motors

ignaciobfp, 2018-12-17:
### Feature request

**Is your feature request related to a problem? Please describe.**
This parameter is meant to give differential trust on planes with more than 2 motors and insufficient or no rudder control. In a quadplane, this parameter has no effect when motors are properly configured i.e. servo5function=33 (QuadPlaneMotor1) and servo6_function=34 (QuadPlaneMotor2).

As a workaround for this, front motor functions can be set as 73 (ThrottleLeft) & 74 (ThrottleRight) respectively, then this feature works as expected in FBWB/A modes but quadcopter modes are broken (front motors does not spin, as it is expected with this setting)

**Describe the solution you'd like**
Motors should behave the same as when they are configured as ThrottleLeft/Right when in plane modes, while retaining its expected configuration in quadcopter mode. Since I assume most setups use close to default motor arrangement, I think a mask/direction control is not a neccesary requirement (on the short term) for this to be implemented.

**Describe alternatives you've considered**
Probably best alternative is to use bigger or more effective control surfaces on the rudder that does not require assistance from motors, but this is not possible in many cases. I think there is not an alternative at the moment, except probably an overkill solution intervening motor controls when in plane modes with an arduino or similar reading rudder from rx and changing the PWM signal that arrives to the ESC if needed.

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Additional context**
The behavior described below was tested on a V tail VTOL plane with 2 vectored motors. It flies great on quadcopter modes, but when using plane modes the tail angle is a bit too wide, so sometimes rudder control is insufficient. The same problem I am facing can be seen on some other user's post like this one: https://discuss.ardupilot.org/t/can-not-have-a-thrust-differential-in-after-transition-tofbwa/32436

I will be glad to provide more information and perform any test that is within my reach.

IamPete1, 2018-12-30:
@ignaciobfp I have just done a PR for a fix for this, It would be great if you could test it, I can build the code for you, what flight controller are you using? 

Thanks

ignaciobfp, 2018-12-30:
@IamPete1 many thanks. The board I am using is an standard pixhawk (FMUv2), It will be nice if you can build the code for me, I will test it as soon as possible.

IamPete1, 2018-12-30:
@ignaciobfp  Great, I have uploaded the .apj [here.](https://1drv.ms/u/s!Apevk2F1cJOqiVmdvpYhTWTMvphU)

Thanks

ignaciobfp, 2018-12-30:
Hello, and thank you one more time for taking the time to fix this.

Tested your build today on the plane, everything works as expected on Q_ modes and on manual plane mode now differential thrust works and rudder control is much better. 

One problem I found is that FWBA mode seems broken, since motors do not appear to transition properly to horizontal position. Instead, they aim downwards like in copter mode, going down gradually as I increase throttle, and returning to vertical position when I lower it. One time they stayed at full throttle until I switched modes again. I am not sure if this is a consequence of this PR or maybe of other changes in master from the version I had (3.9.4, latest stable at the moment).

Can you please check it? I am uploading a video showing the actual behaviour. I did not have the courage to test FWBA in flight, It might be related to GPS speed being 0 instead, but it works on 3.9.4 as-is.

IamPete1, 2018-12-30:
@ignaciobfp glad to hear its working, in part anyway. I hope the FBWA issue it to do with transitions, as you say its probably waiting to reach the transition airspeed.  I will have a go a SITL in the morning and see if i can reproduce. A way to narrow it down would be to arm straight into FBWA or switch from manual into FBWA, this will bypass the transition code. A vid would be great and the log too if possible. 

The other thing check would be if it does the same thing on [master](http://firmware.ardupilot.org/Plane/latest/), maybe something else has changed since stable. 

It is quite possible I have broken something, i'm not that familiar with the tilt-rotor code. Thanks for testing!

ignaciobfp, 2018-12-31:
@IamPete1 thank you for taking the time to test this. In this case this issue can be closed, as it is fixed and unrelated to the bug we are seeing. I will open another one for the transition failing. Thanks a lot.

IamPete1, 2019-01-01:
Lets wait till the fix makes it into master before we close the issue.

IamPete1, 2019-01-14:
Has now been merged

closed 2019-01-14 by IamPete1