ArduPilot/ardupilot #10040: Plane: RC_OPTION to include "Disable Airspeed"

Naterater, 2018-12-17:
### Feature request

With all of the problems encountered when an airspeed sensor fails, users are only left with 1 choice - Manual Mode (because every stabilized mode has airspeed affecting the PID scaling on control surfaces). Let's allow another safer option.

**Describe the solution you'd like**
A simple, yet effective solution, would be to provide a manually activated airspeed failsafe switch on the RC transmitter that effectively changes ARSPD_USE = 0. This falls back on throttle mode and can yield a safe and successful return or completion of mission. This requires the correct tuning of TRIM_THR and TRIM_ARSPD_CM, but that can be noted in the documentation. It's likely far faster and more reliable than any other method.

**Describe alternatives you've considered**
1. Auto-detect the bad airspeed readings: https://github.com/ArduPilot/ardupilot/issues/9372
2. Change the parameter via the GCS. Unfortunately that's not always an option when planes try to immediately rocket into the ground upon airspeed loss.
3. Change TECS to stop continuing to demand dives when airspeed and altitude are too low but throttle is high.

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Additional context**
This conversation has been going on for too long for nothing to be done. I think this is the simplest patch to improve safety.

#1215
#1628
#2787
#3159
#3188
#5951
#4122
#7738
#9372

pompecukor, 2018-12-17:
Nate, actually I do not agree with this one. Simple reason being this is such a rear use case, that no one can be expected to remember that an RC switch is set for this. I wont remeber which is was, let alone a client that has never used it.

The problem should be addressed, yes. But in my many years of UAV building, between me and all my customers, there have been 3 instances where  this would have been worth having and out of those 3, one was me, and as soon as I saw what was happening I switched to FBWA and brought here down (yes there was an effect on PID, but nothing even remotely close to unflyable). The other two occasions, was with customer and none of them realized what the problem was actually, till I went over the logs.
I mean that the ultimate reason for the crash was the AS. One just know it was because of the rain, but nothing more specific. The other was an even more complex situation. 

All in all I don't see this as something useful. It has to be "intelligent". If anyone has the mind set to remember a switch for something that may never ever happen in his/hes carrier. Then you might as well expect that person to be able to open parameters and set use AS to 0 and hit write. It does not take that long.

So let us push for something that really helps. Now I start to wonder how many times did you get a problem from AS that you feel you would profit from getting this on a switch. It is really not a regular thing to happen. Maybe it is the AS sensor that you use???

Naterater, 2019-01-04:
If DISABLE GPS is an option, DISABLE AIRSPEED should be too.

StonedDawg, 2021-06-15:
i would really love to see this happened, since relying on "computer" to detect if there is a hardware failure or not is not really a "failsafe". I have an airspeed sensor but i have never activate the USE_AIRSPEED since there would be no turning it off midflight in case something happened.

peterbarker, 2021-06-17:
See https://github.com/ArduPilot/ardupilot/pull/17790

PR #17790 (2021-06-23): Airspeed disable on-a-switch
Closes #10040
closed 2021-06-23 by peterbarker