ArduPilot/ardupilot #10050: SITL: add correct battery estimator from vehicle type

khancyr, 2018-12-18:
### Feature request

**Is your feature request related to a problem? Please describe.**
Following https://github.com/ArduPilot/ardupilot/pull/9900, we should create a correct way to implement battery consumption on SITL : 
Maybe with a battery object in SITL and with model of consumption on each vehicle backend.

**Platform**
[ X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

ongspxm, 2019-01-26:
i would like work on this, is this still required?

From the code change in #9900 on `libraries/AP_HAL_SITL/SITL_State.cpp` and recommended approach for this, we should have a battery object that returns right throttle value dependent on the vehicle type.

Did i get the approach right?

khancyr, 2019-01-28:
hello,
yes this is still require ! We should do it step by step. The first one would be to have battery object in SITL, then each vehicle could implement its correct consumption based on motors output requested.
Then we could improve the battery, adding new chemistry (lipo, NICD, lead battery, etc), custom discharge rate, temperature, etc etc.

ongspxm, 2019-01-31:
Yep, i would like to work on this, i will first add a battery object in SITL.

ongspxm, 2019-02-05:
I have explored creating a `SIM_Battery`, like that of `SIM_Sprayer` and `SIM_Parachute`.

But the function to calculate the the servo voltage (`float get_servo_voltage()`), I have plans to make it a function in `SIM_Airplane` that can be overriden.

Is this the right track?

ongspxm, 2019-02-06:
Hi, can you help me take a look at #10425 ?

i have trouble overriding the `servo_voltage_and_current` for the rover, But it seems like the `sitl_model` is of type `SIM_Aircraft`, how to access `SimRover`?

ongspxm, 2019-03-04:
pr open

ongspxm, 2019-03-07:
@khancyr any update for this?

ongspxm, 2019-03-13:
haha, any updates?

hendjoshsr71, 2021-09-13:
@khancyr should this one still be open? There is a sim_battery now. But I think there is still work to be done for vehicle type's other than multicopter?

khancyr, 2021-09-13:
We got a proper SITL backend now, but it would deserve some work to be adapted to each vehicles type.
So I think we could leave it open as that a good issue for newcomeers

hendjoshsr71, 2021-09-14:
Great! So as an update the things that still need to done for this issue:

Add the battery backend and an appropriate model to the following using `SIM_MultiCopter.cpp` as a guide
- SIM_Plane
- SIM_Quadplane
- SIM_Rover
- SIM_Submarine
- SIM_Helicopter
- SIM_Blimp

Badrivishal, 2023-03-19:
I have made the changes for SIM_Plane
here is the pull request,
#23256 
Please guide me if I am going in the correct direction.

peterbarker, 2024-09-25:
@hendjoshsr71 could you evaluate this, please?