ArduPilot/ardupilot #10054: Arducopter on Linux and RPI0

JAR4x4, 2018-12-18:
Hello,

I have built a custom FC based on the PXFMINI schematics. The FC is connected to a RPI0 running Raspbian Strech. I have tried both kernel 4.9 and 4.14. The FC works fine until i connect my PCA9586 breakout board. Arducopter runs fine without crash, but I'm not able to connect my GCS via UDP. If i run without the PCA9586 I'm able to connect. GDB back trace reports the following after killing Arducopter with ctrl+c:
```

(gdb) run
Starting program: /home/pi/arducopter -A udp:192.168.1.15:14550
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
Raspberry Pi Zero Rev 1.3. (intern: 1)
[New Thread 0xb6c31450 (LWP 635)]
[New Thread 0xb69ff450 (LWP 636)]
[New Thread 0xb68ff450 (LWP 637)]
[New Thread 0xb66ff450 (LWP 638)]
[New Thread 0xb65ff450 (LWP 639)]
MS5611 found on bus 0 address 0x00
[New Thread 0xb65bf450 (LWP 640)]
[New Thread 0xb657f450 (LWP 641)]
MPU: temp reset IMU[0] 7426 0
^C
Thread 1 "arducopter" received signal SIGINT, Interrupt.
0xb6cfb8ac in ioctl () at ../sysdeps/unix/syscall-template.S:84
84	../sysdeps/unix/syscall-template.S: No such file or directory.
(gdb) bt
#0  0xb6cfb8ac in ioctl () at ../sysdeps/unix/syscall-template.S:84
#1  0x000b8d10 in Linux::I2CDevice::transfer(unsigned char const*, unsigned int, unsigned char*, unsigned int) ()
#2  0x000ba8c8 in Linux::RCOutput_PCA9685::push() ()
#3  0x00000002 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb

```

I'm not that good with the code yet, any suggestions to how this can be fixed?

Alex

patrickelectric, 2020-10-31:
this should be fixed in the last master branch

IamPete1, 2021-08-29:
@patrickelectric say this is fixed so I'm going to close.

closed 2021-08-29 by IamPete1