ArduPilot/ardupilot #10059: Vagrant is not working

eagleman111, 2018-12-19:
### Bug report

I cloned latest ardupilot code, and tried to run vagrant, but it seems not working, this is the output
{
vagrant up
Bringing machine 'artful32' up with 'virtualbox' provider...
==> artful32: Box 'ubuntu/artful32' could not be found. Attempting to find and install...
    artful32: Box Provider: virtualbox
    artful32: Box Version: >= 0
==> artful32: Loading metadata for box 'ubuntu/artful32'
    artful32: URL: https://vagrantcloud.com/ubuntu/artful32
==> artful32: Adding box 'ubuntu/artful32' (v20180718.0.0) for provider: virtualbox
    artful32: Downloading: https://vagrantcloud.com/ubuntu/boxes/artful32/versions/20180718.0.0/providers/virtualbox.box
    artful32: Download redirected to host: cloud-images.ubuntu.com
    artful32:
An error occurred while downloading the remote file. The error
message, if any, is reproduced below. Please fix this error and try
again.

The requested URL returned error: 404 Not Found
}

**Version**
Latest

peterbarker, 2018-12-19:
On Wed, 19 Dec 2018, eagleman111 wrote:

> I cloned latest ardupilot code, and tried to run vagrant, but it seems not working, this is the output

Please paste the output of "git show"

eagleman111, 2018-12-19:
git show
commit cb3b0bd37398ba66a99723337d87401f667fec8e (HEAD -> master, origin/master, origin/HEAD)
Author: Andrew Tridgell <andrew@tridgell.net>
Date:   Sat Dec 8 12:36:30 2018 +1100

    SITL: don't try to connect Morse sockets so quickly

diff --git a/libraries/SITL/SIM_Morse.cpp b/libraries/SITL/SIM_Morse.cpp
index e27c17c52..b9bb5d4ab 100644
--- a/libraries/SITL/SIM_Morse.cpp
+++ b/libraries/SITL/SIM_Morse.cpp
@@ -262,7 +262,8 @@ bool Morse::connect_sockets(void)
             AP_HAL::panic("Out of memory for sensors socket");
         }
         if (!sensors_sock->connect(morse_ip, morse_sensors_port)) {
-            if (connect_counter++ == 1000) {
+            usleep(100000);
+            if (connect_counter++ == 20) {
                 printf("Waiting to connect to sensors control on %s:%u\n",
                        morse_ip, morse_sensors_port);
                 connect_counter = 0;
@@ -280,7 +281,8 @@ bool Morse::connect_sockets(void)
             AP_HAL::panic("Out of memory for control socket");
         }
         if (!control_sock->connect(morse_ip, morse_control_port)) {
-            if (connect_counter++ == 1000) {
+            usleep(100000);
+            if (connect_counter++ == 20) {
                 printf("Waiting to connect to control control on %s:%u\n",
                        morse_ip, morse_control_port);
:...skipping...
commit cb3b0bd37398ba66a99723337d87401f667fec8e (HEAD -> master, origin/master, origin/HEAD)
Author: Andrew Tridgell <andrew@tridgell.net>
Date:   Sat Dec 8 12:36:30 2018 +1100

    SITL: don't try to connect Morse sockets so quickly

diff --git a/libraries/SITL/SIM_Morse.cpp b/libraries/SITL/SIM_Morse.cpp
index e27c17c52..b9bb5d4ab 100644
--- a/libraries/SITL/SIM_Morse.cpp
+++ b/libraries/SITL/SIM_Morse.cpp
@@ -262,7 +262,8 @@ bool Morse::connect_sockets(void)
             AP_HAL::panic("Out of memory for sensors socket");
         }
         if (!sensors_sock->connect(morse_ip, morse_sensors_port)) {
-            if (connect_counter++ == 1000) {
+            usleep(100000);
+            if (connect_counter++ == 20) {
                 printf("Waiting to connect to sensors control on %s:%u\n",
                        morse_ip, morse_sensors_port);
                 connect_counter = 0;
@@ -280,7 +281,8 @@ bool Morse::connect_sockets(void)
             AP_HAL::panic("Out of memory for control socket");
         }
         if (!control_sock->connect(morse_ip, morse_control_port)) {
-            if (connect_counter++ == 1000) {
+            usleep(100000);
+            if (connect_counter++ == 20) {
                 printf("Waiting to connect to control control on %s:%u\n",
                        morse_ip, morse_control_port);
                 connect_counter = 0;
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
(END)

peterbarker, 2018-12-19:
This should be fixed on master.

WickedShell, 2019-04-09:
Closing as @peterbarker believe this to be fixed in master, and no one has come back either way. If this was wrong please comment on it, and we can reopen as needed.

closed 2019-04-09 by WickedShell