ArduPilot/ardupilot #10068: Quadplane motors spin up when connecting power

blyterate, 2018-12-19:
#### Issue details
_Please describe the problem, or desired feature_

#### Version
_What version was the issue encountered with_

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_

#### Logs
_Please provide a link to any relevant logs that show the issue_

Naterater, 2018-12-19:
We really need logs to re-create your issue. Have you calibrated your ESC's recently? Does the motor RPM change during boot? Is it idle speed or full speed? How long does it last?

lowlyelevated, 2018-12-20:
I've seen an issue where all of my servos (connected to the cube 2.1 main out) move on boot. I actually killed two quadplane tilt servos this week because of it. Unfortunately, the logs aren't very helpful, because the behavior begins before logging even initializes. 
I've never had motors spin up, but my motors are connected to the aux out ports on the cube 2.1. So maybe it only affects the main out ports? Or maybe your ESCs are responding the the erroneous input differently than mine are. (I'm using flyfun 40a units) 

I'm just spitballing here, I've honestly got no clue what's causing it. this old Discuss thread may be of interest? https://discuss.ardupilot.org/t/servos-at-extreme-positions-during-boot/20296/30

WickedShell, 2019-04-09:
Was this ChibiOS or PX4 based builds? I've seen this under PX4 builds, and the root problem there is fixed under the ChibiOS ones.

IamPete1, 2021-07-26:
@WickedShell says this is fixed on ChibiOS

closed 2021-07-26 by IamPete1