ArduPilot/ardupilot #10072: Copter: throttle spike if user switches from Auto takeoff to Loiter

rmackay9, 2018-12-20:
I've found a significant throttle output spike when switching from Auto to Loiter.  To recreate the issue do this:

- load up a simple mission with a takeoff to 20m
- arm the vehicle (in Loiter), switch to Auto and raise throttle to begin the mission
- before the vehicle reaches 20m, switch to Loiter

[image]

SergeyBokhantsev, 2018-12-20:
Could that be an inverted case of [this bug](https://discuss.ardupilot.org/t/throttle-drop-when-switching-to-loiter-from-rtl/34388)?

rmackay9, 2018-12-20:
@SergeyBokhantsev,

Yes, it could be the same as this issue: https://github.com/ArduPilot/ardupilot/issues/7435
Also @khancyr has proposed a fix for the issue here: https://github.com/ArduPilot/ardupilot/pull/9858

khancyr, 2018-12-20:
@rmackay9, It could come from the same root. But on my fix I remove the pos controller reset part.
I will test with the pos controller reset but it is a 100ms matter. So it should have another cause

lthall, 2021-09-27:
All this code has been rewritten in 4.1 with more stringent initialisation.

If this is still a problem we can reopen.

closed 2021-09-27 by lthall