ArduPilot/ardupilot #10084: Feature request: support SmartAudio VTX control protocol

macrokernel, 2018-12-21:
Please implement support for the SmartAudio VTX control protocol to allow VTX power level, band and channel programmatically.

Many modern VTX models allow changing their configuration parameters via an additional wire connected to the FC TX pin. This nice feature is implemented in BetaFlight and iNav. There is a good description of how it can be used in the Oscar's blog: https://oscarliang.com/vtx-control/

I think that it would even better if the VTX power level could be changed on arming. For example, when the model is armed, VTX is started with the minimal power level to prevent overheating. After arming the model, VTX power could be set to its maximum or another configured value.

#### Platform
[  ] All
[  ] AntennaTracker
[ * ] Copter
[ * ] Plane
[ * ] Rover
[ * ] Submarine

ThisIsTheOnlyUsernameAvailable, 2018-12-22:
There is some initial work done by Sergey here: https://github.com/SergeyBokhantsev/ardupilot/commits/SmartAudio_on_3.6

It implements the basics so would be a good starting point for anyone considering the project.

SpaceGoast, 2019-05-07:
I posted on ardupilot that I'm willing to put $20 through paypal to implement this. Hoping others will join
[](url)https://discuss.ardupilot.org/t/is-there-a-way-to-get-smartaudio-working/29025/12

mric3412, 2019-05-19:
Great idea ! (Pitmode before arming and preconfigured power level after arming)
To complete this new feature, what about an automatic control of the VTx power as a function of the "distance to home" ?
More simply, add a new parameter "DIST_MAX_VTX_PWR" (in meter) : as soon as distance_to_home > DIST_MAX_VTX_PWR, the flight controler sends to the VTx (via smartaudio) the message for switching to max power

whoim2, 2019-12-11:
Good idea - change vtx power via RC chan value, for use aux or 3-pos switch for this.

jimpriest, 2019-12-17:
+1 on this.  As VTX get more powerful - it's nice to power on in pit mode and then go full power on arm.

brucewillis88, 2020-01-29:
I would also like to see this and I would specifically donate for this feature

iromadm, 2020-10-11:
We need that!

closed 2021-02-17 by andyp1per