ArduPilot/ardupilot #10094: Dataflash Log Auto Analysis updates for Copter-3.6

rmackay9, 2018-12-22:
I think I've reported this before but when running a Copter-3.6 binary log through the MP's auto analysis tool (which I think is just taken from this script) we get the following incorrect outputs:

- Test: NaNs = FAIL - Found NaN in POS.RelOriginAlt, Found NaN in CTUN.TAlt, Found NaN in CTUN.DSAlt
- Test: Parameters = FAIL - 'THR_MIN' not found
- Test: Pitch/Roll = UNKNOWN - 'BarAlt'

WickedShell, 2018-12-22:
@rmackay9 surely this is a MP issue rather then an ArduPilot one, unless it's relying upon some internal tool? (I thought MP had it's own tool for that though).

rmackay9, 2018-12-22:
@WickedShell, I originally raised the issue in the MP issues list but it seems like MP is just using [this script](https://github.com/ArduPilot/ardupilot/tree/master/Tools/LogAnalyzer) which we maintain.

WickedShell, 2018-12-22:
@rmackay9 ah okay then definitely relevant. Although this appears to duplicate #9922 so maybe we should close one of them? (Happy to see it be tracked here now if this is better)

rmackay9, 2018-12-22:
I ran the script myself manually on an ubuntu box against a Copter-3.6.4-rc1 log ([here](https://www.dropbox.com/s/v2p19kkuj1i0fwu/test.log?dl=0)) and received the following output which indicates that the THR_MIN issue is fixed but the NaN issue isn't completely resolved (still exists in POS.RelOriginAlt and CTUN.TAlt)

> Unknown mode=0
> Unknown mode=5
> Unknown mode=0
> Unknown mode=5
> Unknown mode=5
> Unknown mode=5
> Unknown mode=3
> Unknown mode=5
> Unknown mode=5
> Unknown mode=3
> Unknown mode=5
> Unknown mode=3
> Unknown mode=5
> Unknown mode=3
> Unknown mode=5
> Unknown mode=0
> Dataflash log analysis report for file: ../../../GitHub/ardupilot/test.log
> Log size: 12.45mb (155486 lines)
> Log duration: 0:06:35
> 
> Vehicle Type: ArduCopter
> Firmware Version: V3.6.4-rc1 (6011515d)
> Hardware: 
> Free RAM: 0
> 
> 
> Test Results:
>               Brownout:  GOOD                                                              
>            OpticalFlow:  GOOD       Set FLOW_FXSCALER to -100                              
>                                     Set FLOW_FYSCALER to -170
>                                     
>                                     Cal plots saved to flow_calibration.pdf
>                                     Cal parameters saved to flow_calibration.param
>                                     
>                                     FLOW_FXSCALER 1STD = 35
>                                     FLOW_FYSCALER 1STD = 20
>                 Thrust:  GOOD                                                              
>                  Empty:  GOOD                                                              
>                    VCC:  UNKNOWN    No CURR log data                                       
>                Compass:  GOOD       mag_field interference within limits (15.88%)          
>         Event/Failsafe:  GOOD                                                              
>           IMU Mismatch:  GOOD       (Mismatch: 0.23, WARN: 0.75, FAIL: 1.50)               
>             Parameters:  GOOD                                                              
>          Motor Balance:  GOOD       Motor channel averages = [1484, 1483, 1537, 1527]      
>                                     Average motor output = 1507
>                                     Difference between min and max motor averages = 54
>              Vibration:  UNKNOWN    No stable LOITER log data found                        
>          Dupe Log Data:  GOOD                                                              
>                   NaNs:  FAIL       Found NaN in POS.RelOriginAlt                              [GRAPH]
>                                     Found NaN in CTUN.TAlt
>                    GPS:  WARN       Min satellites: 5, Max HDop: 2.12                          [GRAPH]
>             Pitch/Roll:  GOOD                                                              
>                     PM:  GOOD                                                              
>               Autotune:  UNKNOWN    No ATUN log data                                       
> 
> 
> The Log Analyzer is currently BETA code.
> For any support or feedback on the log analyzer please email Andrew Chapman (amchapman@gmail.com)
> 
>

peterbarker, 2024-08-20:
We've remove the LogAnalyzer now; the web-based tools are replacing it.

closed 2024-08-20 by peterbarker