ArduPilot/ardupilot #10095: UAVCAN Compass: appearing as compass 2

rmackay9, 2018-12-22:
We have a report in Copter-3.6 beta testing in which a user has 2 UAVCAN GPS/Compass module's attached.  There are two slightly unusual things happening:

- this message appears twice in the logs "GPS 1: specified as UAVCAN".  I would expect the 2nd time it appears to be "GPS 2:"
- the UAVCAN compass(s) are appearing as COMPASS3.  They are external so I would expect them to appear as the first compasses in the list.

[This is the discussion where the issue appeared](https://discuss.ardupilot.org/t/everything-solved-all-great-now/36480).  The user has since deleted the log but I have a copy.

EShamaev, 2018-12-22:
As for GPS, I have a plane with 2 UAVCAN GPS and they are enumerated and used correctly with blending. Can you please send me logs and version of firmware that was used.

As for compasses appearing at the end of the list - that is true.
UAVCAN compasses are enumerated after all other compasses, so they get added the last. This was discussed during implementation.

EShamaev, 2018-12-22:
I could not recreate the problem.

According to log, there are severe problems with GPS
[image]

For messages, it's even more strange as this message is coded in GPS_Backend.cpp that is common to all GPS's and not UAVCAN specific. That number is taken from state.instance

For example GCS messages from my last log:
MSG, 15980008, GPS 1: specified as UAVCAN
MSG, 15980063, GPS 2: specified as UAVCAN

There might be a chance that both GPS's have same node ID on CAN bus, but that is shown only in terminal messages.

My recommendation is to check node ID on both GPS and find out what is going on with first GPS receiver and if the problem persists, evaluate further with logs from the terminal.

EShamaev, 2018-12-22:
The situation on compasses needs to be discussed again, however.
Now being external, the UAVCAN compasses are still enumerated in the very end.

I was already thinking what can can be done as:
- number of compasses limited to 3 is not enough already in many use cases
- compass identification is not ideal
- calibration of compasses for large frames is painful

I would be glad for any input about this

rmackay9, 2018-12-22:
@EShamaev, I think we should talk about this perhaps with @Tridge on or before the next dev call.

For the last point we've got the new inflight compass calibration method, it just needs to be documented.

highfreq, 2018-12-24:
Have to find what is the hardware of the second on board compass. I know one is the same hardware as the external so i can't disable the driver, otherwise also the external would not work.
Probably disabling the second on board will make space for the second UAVCAN.
Now i need to understand what is the hardware i want to disable.

highfreq, 2018-12-24:
Oh, almost forgot, thanks for your support and Merry Christmas.

Naterater, 2019-01-08:
For what it's worth, it appears that compass 3 z-values do not keep the same trends that the other compasses do. The X-Y values are fine (just offset a bit from one another).
[image]

Naterater, 2019-01-08:
X and Y charts here:

[image]

[image]

highfreq, 2019-01-17:
I have read the devcall on the forum and it looks like my log is really scary. I would like to understand what the problem has been and patch it. If somebody feels the UAVCAN gps are the root of the prob i'll take them off.
Even if i am no longer on the Ardupilot Discuss forum, i can still test whatever patch would come out and feed any kind of info that may be needed.

Corrado

IamPete1, 2022-09-11:
No new reports, this area of the code has changed significantly, so unfortunately I don't think there is much more we can learn from this issue. I think we should close, @rmackay9?