ArduPilot/ardupilot #10096: Antenna Tracker chibios builds feature request

hdtechk, 2018-12-22:
Issue details. No just features 
Since nuttx is being Removed. Is it possible to start a new beta version of Antenna Tracker. Based on chibios.   There has also been a few commits to tracker master since Aug 2016 latest stable of tracker 

#### Version
_What version was the issue encountered with_

#### Platform
[  ] All
[ x ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type

#### Hardware type
Could add lots of new boards for tracker

#### Logs
Not relevant

rmackay9, 2018-12-22:
I think we need to get @peterbarker an antenna tracker so he can take over as maintainer.. I'll raise a proposal for AP to buy him one.

IamPete1, 2021-07-26:
We now build tracker for everything, https://firmware.ardupilot.org/AntennaTracker/latest/

closed 2021-07-26 by IamPete1