ArduPilot/ardupilot #10103: RunCam Split Mini in px4?

manotroll, 2018-12-23:
is it possible to use the RunCam Split Mini in px4?
How do I make her serial connection on px4?
it can be shared with the miniosd since runcan would practically not receive external commands
How do other open-source controllers already support RunCam not add this support to px4?

RunCam Support (Customer Services)

Dec 22, 10:08 CST
Sorry for that Split mini doesn’t support PWM control.

Regards,
Stella T.
RunCam Customer Service

as it does not support pwm control would not it be possible to add its support?
if it was compatible many would use it I already think that the controllers that supports px4 have more serial than others

https://discuss.ardupilot.org/t/runcam-split-mini-in-px4/36625

auturgy, 2018-12-23:
https://support.runcam.com/hc/en-us/articles/115011786628-RunCam-Split-Communication-protocol?mobile_site=true

rmackay9, 2018-12-24:
I think this request is to add support for controlling the RunCam using a UART.  I don't actually know what functions are supported using the UART but here is a link to the runcam website: https://www.runcam.com/

And here is a picture of the rear side of the camera showing the pins.

[image]

I've edited the subject line and description slightly to remove some misleading words.

auturgy, 2018-12-24:
My link above has details of the protocol

manotroll, 2018-12-24:
Thank you for reformulating
I already asked him and the mini does not support commands by pwm

atshort, 2018-12-25:
In addition to the protocol there's the question of the user interface. Would be idea be to have ArduPilot use the same stick commands for menu navigation as [Betaflight](https://github.com/betaflight/betaflight/wiki/RunCam-Device-Protocol) uses for RunCam serial devices?

manotroll, 2018-12-25:
I do not know how and the interface in Betaflight could gain a tab on the optional hardware

gnumarco, 2019-05-14:
I have a RunCam Swift 3, seems it can be controlled using a UART and the controller's sticks, but it works only in betaflight. It would be great to have support for this !

mikerob, 2019-05-14:
Here's the Betaflight pull request - their implementation in the FC and a link to the actual protocol definition.

https://github.com/betaflight/betaflight/pull/4251

The GCS interface is a problem, but given that there are existing camera control messages and functionality, it seems like it wouldn't be too hard to implement it as an option at least for the camera trigger / start and stop recording functions that already exist in MP / Ardupilot.

gnumarco, 2019-05-14:
If I knew how to code for PixHawk, I'd do it, but I have no idea :(
But it would be a great feature to have, definitely!

manotroll, 2019-05-14:
this is good that a developer has made available to put the feature
then I would have the smart audio plus this is something else
support could be for all RunCam

gnumarco, 2019-05-15:
I have read the Betaflight pull request and the documentation, then I have gone a bit trough the ArduPilot dev doc, but I am afraid that, although I am a professional computer scientist, I am too used to Python and Java to efficiently implement this :(

nwinova, 2019-07-24:
Does this mean there is no way to use the digital camera control of the swift 3 v2 with a kiss flight controller?

Inv3nt0r1, 2019-10-25:
Is there any work going on to add this feature?

rmackay9, 2019-10-26:
@Inv3nt0r1, I haven't seen any developer working on this but it's sometimes hard to know for sure.

andyp1per, 2019-10-26:
I now have a runcam so plan to look at it in due course.

Andy

manotroll, 2019-10-26:
I could not import the code in inav?
after you implement support could you try looking at smartaudio?

andyp1per, 2019-11-10:
I have started on this. Still a work in progress.

https://github.com/ArduPilot/ardupilot/pull/12810

andyp1per, 2020-01-01:
This has been fixed and is in master

closed 2020-01-01 by andyp1per