ArduPilot/ardupilot #10105: Kakute F7 Arducopter build, ChibiOS, PX4Flow not recognized.

markandkymward, 2018-12-23:
### Bug report

**Issue details**

When building firmware for Holybro Kakute F7 with an attached PX4Flow the optical flow sensor was not recognized. 

A work around is to insert the line 

backend  = AP_OpticalFlow_PX4Flow::detect(*this)
in line 121 of ardupilot/libraries/AP_OpticalFlow/OpticalFlow.cpp of the current build.

This fix is insufficient for master release and should be addressed.  I noticed that the Kakute is not defined in ardupilot/libraries/AP_HAL/AP_HAL_boards.h

In addition, use of the default address of 0x42 may not work in all builds.  I needed change it to 0x45 by soldering appropriate jumpers and providing appropriate address offset of 3 in Mission Planner before achieving consistent results.

**Version**
3.7-dev, ChibiOS.

**Platform**
[  ] All
[  ] AntennaTracker
[x  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quadcopter

**Hardware type**
Kakute F7, PX4Flow v1.3.1

Naterater, 2019-01-31:
[image]
Verified by 
[image]

magicrub, 2019-01-31:
here's a direct link to that FB post:
https://www.facebook.com/groups/ArduPilot.org/permalink/2202053956484452/

rmackay9, 2020-01-13:
Here is a topic on the forum with the same complaint: https://discuss.ardupilot.org/t/kakute-f7-mini-and-kakute-f7-aio-px4flow-not-working-in-copter-4-0/51000