ArduPilot/ardupilot #10117: Copter: support horizontal thruster motors

rmackay9, 2018-12-26:
Support adding motors to provide additional horizontal thrust to allow the vehicle to move horizontally without leaning.

P.S. I thought we had an issue here in the issues list for this request but I can't find.

auturgy, 2018-12-26:
Is there a specific use case? It can already be done in plane via qassist

rmackay9, 2018-12-26:
@auturgy, This is apparently helpful in cases where a large camera is intended to always be downward facing but is too large to be easily mounted on a 2-axis or 3-axis gimbal.  There was a long justification about whether it is more efficient and the conclusion (from my point of view in any case) was that it is actually less efficient.

IamPete1, 2018-12-26:
As mentioned plane does this already, rather than using Q-assist I would just set he transition speed very high so it can never transition to a forward flight mode. The booster motor gain is controlled by the  Q_VFWD_GAIN parameter.

I think this brings up the slow converging of the codes. Plane is doing more and more things that copter has and this would start bringing plane things to copter. Some big merging of codes would be a huge job, maybe it is better to just keep duplicating functionality for the time being.

Naterater, 2018-12-27:
Pythagorean theorem proves that 2 motors 90-degrees from each other is less efficient. A mode with a constant pitch/roll angle might be worth it. I guess I don't see the practical gain of having constant lean angle for any application. Instead, mount your camera at a 15-degree pitch up when the craft is neutral so that the camera is NADIR during forward flight.

bnsgeyer, 2019-02-13:
I have been working on a 3.7 version of compound heli which uses two forward facing tailrotors to not only provide directional control and torque balance but also provide thrust for high speed flight. Haven’t figured out whether this will be more efficient but that is part of why I’m doing it. 
I had this type of heli flying for the better part of 2 years before I had a drivetrain failure and it crashed. I got the electronic mixing to work for manual modes but never implemented a control scheme for auto or loiter modes. 
I started working on a way to incorporate the thruster into the position controller. So I made a compound heli model in real flight which lets me test my thruster control logic. I got side tracked with some other stuff and hadn’t got back to it. 
Here is the link to my branch. 
https://github.com/bnsgeyer/ardupilot/commits/compound_heli_3.7

lvale, 2019-02-13:
> There was a long justification somewhere about whether it is more efficient to use these horizontal booster motors to control position rather than leaning and the conclusion (from my point of view) was that it is actually less efficient to have horizontal booster motors.

so, if we follow this concept to the end (copter and possibility of positioning) we'll end up with a 6DOF vehicle like ArduSub, but with large props.....

[image]

bnsgeyer, 2019-02-13:
For Heli's, my intent is to only use forward thrust.  
My general design concept is to have an RC aux function available for forward thrust which is mapped to the motors library, forward_in, variable.  that is very straight forward and easy to do.  the harder part is how to integrate it non-manual modes.  for any mode where the pitch and roll stick inputs from the user command pitch and roll attitude or rate, then the forward thruster is controlled by the RC aux input channel.  I have decided to integrate the thruster into the position controller for modes where either the user pitch and roll input commands speed or auto/guided mode.  Right now I just have the accel to lean angle method pushing a command to the forward thruster instead of pitching the aircraft for x acceleration.  I will need to add a PID for the forward thrust to close the loop on acceleration.  I may do some sort of sharing between lean angle and forward thruster for accel where the forward thruster is only used to bring pitch attitude back to zero over the long term.

SwiftGust, 2020-04-22:
@bnsgeyer Few years back, I did work on small compound helicopter using coax-copter frame
and there is failed attempt PR of my very primitive code https://github.com/ArduPilot/ardupilot/pull/6529
back then I enabled use of forward thruster with AC_PosControl inherited AC_PosControl_Compound class adding proportional controller for the single forward thruster.

With that code I was able to fly compound helicopter in takeoff -> waypoint navs -> land in Auto mode.
This primitive code maybe helpful on non-maual control modes integration.
https://github.com/SwiftGust/ardupilot/blob/354d6383a75a8e1c71739f1654ab802bd986ec5b/libraries/AC_AttitudeControl/AC_PosControl_Compound.cpp#L178-L213

bnsgeyer, 2020-04-24:
@SwiftGust  Thanks for pointing me to your method for integration of the thruster to non-manual modes.  Truthfully I've been pretty busy, so I'm not sure when I will get back to this.  I really need to get another compound heli built so I can finish the integration into non-manual modes.

rmackay9, 2021-05-08:
I've re-opened thsi because I don't think we have a general solution to allow users to connect a horizontally facing motor/propeller.  I'm happy to be corrected but if so then we should enhance the wiki to explain how this can be done.

danlavee, 2023-09-20:
I would like to take a shot at this, but I am not familiar with the ArduPilot code.
Would anyone be willing to spare some time for guidance?

I did go through the code and related PR #6529 by @SwiftGust

PR #16105 (2021-02-09): Copter: add 6DoF support
This adds support for copters that can fly in any direction and at any orientation.

https://youtu.be/CXIlZ8mCfGM

This adds a new attitude control and motor back-end and scripting bindings for each. Scripting can be used to set the true vehicle attitude while the existing navigation 'attitude' is used for the thrust vector. Scripting is also used to set the 6DoF matrix for the mixer. I did spend a some time trying to calculate the mixer directly from motor positions and trust vectors, but it all got too complex. The ideal mixer can be found and then loaded with the sc [... cut]