ArduPilot/ardupilot #10121: Copter: add support for ICE engines (especially for TradHeli)

rmackay9, 2018-12-27:
We should add support for starting/stopping internal combustion engines by adding support for the [AP_ICEngine library](https://github.com/ArduPilot/ardupilot/tree/master/libraries/AP_ICEngine).

[A wiki page is being created here](http://ardupilot.org/copter/docs/common-ice.html).

ChristopherOlson, 2018-12-27:
This is already supported in TradHeli, due to helicopter engines being throttle controlled with idle setting and cutoff (engine stop). Works with all nitro, gas and turbine engines. Starting of turbines is throttle controlled thru their FADEC. Electric starting of piston is controlled with a separate channel with momentary switch.

rmackay9, 2018-12-27:
@ChristopherOlson,

OK great.  @lvale told me that the setup information is in the last video on [this page](http://ardupilot.org/copter/docs/traditional-helicopter-configuration.html) which is great (many people prefer watching videos to reading) but I would like to get the critical information down in writing - in particular:

- do we connect an ignition and starter to the servo outputs on the flight controller?
- if "yes" how do we configure which outputs or are they hardcoded?
- what's the sequence the pilot goes through to start and stop the engine?

ChristopherOlson, 2018-12-27:
Helicopter gas engines are magneto ignition (same as full-size aircraft engines), with the exception of the Gaui GX-9's F-50, which is CDI. Nitro engines have no ignition system. All turbine engines use a FADEC (Full Authority Digital Engine Control), which controls both ignition and starter and is fully throttle controlled from the RC or flight controller.

Heli's RSC system has support for internal combustion engines built-in with the H_RSC_IDLE for engine idling. Startup is as simple as arming and starting the engine - typically recoil start, but can be electric with handheld starter (quite common with nitros), or throttle start by advancing the throttle to 100% for 1 second, then reducing throttle to ground idle power (turbines).

So there is no need for ignition and starter outputs, except for one outside case - the Gaui GX-9. And there extra channels can be used with servo operated switches or relays, one for the CDI ignition, the other for the starter. Even so, shutdown is automatic by disarming - it is designed into the throttle control. Disarming simply closes the throttle plate all the way so the engine stops.

Anybody flying these more advanced helicopters is already familar with how to start and stop their engine. These machines are not for beginners.

For support of engines in multicopter, however, it is different. Multicopter does not have the HeliRSC.

ChristopherOlson, 2018-12-27:
@rmackay9 for support of combustion engines, there must also be a way to control the engine's power. Again, Heli has this built-in with the throttle curve.

To add this support to multi's there will likely be two types of drives - one that could be driving a generator to charge batteries in flight. The other that could be direct propulsion or thrust, such as the example tricopter with an assist lift rotor or prop. Since the two types of drives are significantly different, one would likely use a governor of some sort, the other maybe a throttle curve of some sort, this would have to be designed for multi. Bits of the code from the HeliRSC could probably be used to start and stop a gas engine on a multi. But things like governors or throttle curves become more complicated as it requires speed sensor and/or tailoring the throttle to thrust as is done with a throttle curve.