ArduPilot/ardupilot #10128: Plane: Enable Manual Servo Outputs while Initializing

Naterater, 2018-12-28:
### Feature request

**Is your feature request related to a problem? Please describe.**
If a flight controller ever happens to reboot for a split-second in the air, the plane is totally doomed, even if RC control and telemetry are available. Servo outputs are not honored while the board is in the INITIALIZING mode, and the board won't initialize until the accelerometers/gyros are consistent enough (still on the ground).

This can be tested during boot as well - RC outputs aren't available until the plane is still enough to exit the INITIALIZING mode.

**Describe the solution you'd like**
Allow servo outputs in manual mode if RC signal is valid (and safety button not enabled).

**Describe alternatives you've considered**
Get another flight controller or find a way to go straight from the RC receiver.
Disable accelerometer/gyro calibration on boot (boat mode I think).

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Additional context**
I have had 2 telemetry ESC's explode on separate occasions in mid-air, and I think that has caused a voltage spike or short that made in-air reboots happen. Without control, the plane crashed.

peterbarker, 2018-12-30:
On Fri, 28 Dec 2018, Nathan E wrote:

> Allow servo outputs in manual mode if RC signal is valid (and safety button not enabled).

Enabling servos in initialising mode is probably not right; what we 
probably need to do is make gyro and accel cals asynchronous (like 
airspeed).

I do like this idea....

magicrub, 2018-12-30:
I agree with Peter. Moving the gyro Cal to an async process is the way to go. There are several changes needed to allow surviving an airborne reboot. This is one step. Another is for the aircraft to know it rebooted while airborne (hard) and it then does things like automatically skip the Cal (use previous, like boat mode) and setting its home location to the previous known home (surely this is very very bad if done unintentionally), including baro alt (also very bad if done unintentionally), and automatically do an RTL back to home.

Naterater, 2018-12-30:
I think enabling servos ONLY for manual mode might just be just an extension of OVERRIDE_CHAN and OVERRIDE_SAFETY and not require the intelligence or effort of a full automated recovery. This is just a request for servo outputs for an in-air reboot so that a temporary autopilot failure does not mean a crash for flights that are line-of-site to the pilots. It at least gives one backup option.

The last post-crash investigation revealed that the disconnect happened during the start of a turn, resulting in a stall-spin scenario (where neutral servos would have at least given a dynamically stable aircraft a chance to return to a level glide).

shellixyz, 2021-07-26:
Looking forward to this being fixed. Happened to me twice.

1. ESC cap failure resulting in a short brownout -> crash and trashed plane
2. Misconfiguration resulting in 100% throttle during an extended period of time during RTL climb causing a brownout -> crash but fortunately the plane crashed not too far away, landed in a soft field and has been recovered, no damage

In both cases the plane could have been saved by restoring manual control

peterbarker, 2021-07-29:
This requires our gyro calibration step to go asynchronous.  It's hairy.

Naterater, 2022-02-15:
Can we please look into this again?

CrunkleFloop, 2023-01-09:
I understand how having servos moving when the gyro is calibrating could be problematic.

However what if the system was set so that if Manual mode is selected whilst the gyro is in it's calibration phase it bypasses the calibration and then will not allow switching out of manual mode until the FC is rebooted and normal calibration can take place?

This would allow the safety feature of Manual mode in the event of an FC brownout but also prevent the switching back into a stablised mode with an uncalibrated gyro.

shellixyz, 2023-01-09:
@timtuxworth It hasn't been a problem in BonafidePirate's video because the MCU didn't completely reset. Looks like the MCU froze, triggering the watchdog. After a watchdog reset the gyro calib is skipped. IMO he has just been extremely lucky. The voltage probably battery probably didn't go low enough for long enough. The problem is when the brownout causes a complete MCU reset so it's doing a "cold start"

Jotham-B, 2023-06-18:
As it turns out, ESC short circuit failures are more common than I initially thought. Our local club lead says he's seen 3 fail this way this year already. Personally I've never had it .. until now. The last thing showing on the OSD is the current exceeding 2x normal max current draw and that's it, end of video. Everything goes out, the radio says "telemetry lost". But then it comes back, "telemetry recovered". At this point the video (DJI V2 goggles) take some time to re-establish a link, but before that can happen, "telemetry lost" plays again. It's hit the mountain side. If we I had manual control without video link, the plane would still be gone. Had I been flying analog, yes, it could have been saved.

What could be done better?

Prevent the brownout: 
  1. Fuse the ESC. An automotive blade type fuse would likely have saved this vehicle. I would choose a value of 2x the ESC current rating.
  2. Run a separate power system for flight critical components. Being a small wing.. the fuse seems like the best option.

Apply software fixes:
  3. Do something to keep us airborne post-reboot (loiter a few circles, then RTL? is this even practical from cold-boot state?).

I think software fixes are a nice-to-have. If the ESC blows up, I think a fuse is the better/easy/KISS solution.