ArduPilot/ardupilot #10134: Parameter issues

ktskt, 2018-12-29:
Dear, developers.

I found that many parameters have some range checking issues.
For example, each controller’s P parameter (e.g., x, y, z-axis and Roll, Pitch, Yaw P parameter of the PID controller) can be an arbitrary number without a proper range check. 
This leads to two types of problems that I would classify.

Issue 1. Users can put any arbitrary values such as minus values or zero, which leads to crash a drone or be stuck at a fixed location.
Issue 2. As you know, the valid ranges of control parameters are different according to the control model.
As an example of iris+ quadcopter model in the simulator, let me talk about x,y-axis velocity controller. 
Some value of the high P parameter even within ranges (e.g., 5.5 or 6 which is within the valid range, 1-6) could lead to unstability, which may lead to crash. 
For Issue 2, I think the valid ranges in the parameter document may be modifed with *tighter* ranges.
As an example of the P parameter for x, y-axis velocity controller, it should be changed from 1-6 to 1-5 or to 1-4.
Range document: http://ardupilot.org/copter/docs/parameters.html .

So… I am wondering 
1) Do you agree that the above issues are really problems (although Issue 2 may be hard to be solved due to the variety of control model)?
2) Do you plan to patch the above problems for both Issue 1 and 2?

Make sure that *at least*, I believe many people agree that -1 for P parameter is not valid.
Also, I found that some patches against similar issues for filt_hz. 
e.g., https://github.com/ArduPilot/ardupilot/blob/master/libraries/AC_PID/AC_PI_2D.cpp#L42
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AC_PID/AC_PI_2D.cpp#L168
Related patch was applied with commit c10b0b34caa50bc58daf7dd717127941d9017737.
I think it must be patched since they are similar cases.

**Version**
Multiple versions including 3.3.3, 3.4.xx, 3.5.7.
I believe, at least, Issue 1 must be applicable to the recent version.

**Platform**
[  ] All
[  ] AntennaTracker
[O] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quadcopter. But, Issue 1 must be applicable to any type

**Hardware type**
Any types

rmackay9, 2018-12-30:
@ktskt, I'm happy to accept PRs into the Copter and Rover code to help protect against clearly incorrect parameter values so if you are thinking of doing some PRs they are welcome!  I'd recommend starting small and easy as a first step.

ktskt, 2019-01-02:
@rmackay9 
To help your understanding, I attached the below picture.
[image]

This picture represents that case (parameter = 5.125 with iris+ SITL model).

ktskt, 2019-01-03:
Ok, I want to work on Issue 1 as a first step (Protection against clearly incorrect value) because new ranges for VELXY_P may not be easy to be applied. 
But, could you confirm that they are bugs or improvement items which must be patched before I start handling the four cases (including the below additional three cases)? 

Otherwise, I am not 100% sure whether I am wasting my time on patching.
Let's start with VELZ_P, ATC_ANG_RLL_P and ATC_ANG_PIT_P since VELXY_P is confirmed.

1. Case VELZ_P = -1
[image]
In this case, Green line = reference and Redline = sensor measured state for z-axis velocity PID controller

2. Case ATC_ANG_RLL_P = -1
[image]
In this case, Green line = reference and Redline = sensor measured state for roll PID controller

3. Case ATC_ANG_PIT_P = -1
[image]
In this case, Green line = reference and Redline = sensor measured state for pitch PID controller

peterbarker, 2019-01-03:
On Thu, 3 Jan 2019, ktskt wrote:

> Ok, I want to work on Issue 1 as a first step (Protection against clearly incorrect value) because new ranges for VELXY_P may not be easy to be applied.
> But, could you confirm that they are bugs or [...]

I'm sure you must have seen our current code for sanity-checking some PID 
values in ArduCopter?

It was one of the pieces of code to go when I was trying to minimise code 
sizes at one stage....

ktskt, 2019-01-03:
@peterbarker  Thank you for your feedback. This is the reason why I tried to receive confirmation from you guys.

As far as I know, I couldn't find such sanity-checking codes. Also, when I put the aforementioned crazy values into respective P parameter, it shows the above *extremely unstable* control performance (Refer to attached pictures)
This indicates that there are missing or buggy parameter sanity-checking logics. 

However, according to your opinions, there are some sanity-checking logics that I couldn't find. 
If so, I can improve such protection logics since ArduCopter still shows the above poor stabilization performance. (I also shared concrete parameters and setting (iris+ quadcopter in SITL). 
I also observed the same results in the real iris+ quadcopter. (Actually, the real copter shows worse performance since it much more easily crashed).
Otherwise, I need to put my codes without scratch.

So, again, (1) I believe the above cases are bugs since they can be triggered via GCS. So... do you agree? and (2) If (1) is true, is there related sanity-checking logic? if so, I can fix that code instead of writing new codes. At least, I couldn't find such codes.

And then ... if the code size is a problem, please give me some feedback when I submit my PR.

peterbarker, 2019-01-04:
https://github.com/ardupilot/ardupilot/blob/master/ArduCopter/AP_Arming.cpp#L234 - perhaps @ChrisBird might have some input here as he contributed those lines.

Only checks against zero values.  Any other (non-negative!) value is
frame-specific ('though I'd guess non-physically likely!).

Since you're looking at this anyway - can you say that those `is_zero(...)`'s in those existing lines should be `is_positive` instead?

I'm wondering if we could have an "is_valid(...)" call on the PID object directly - but I've got a feeling that it might be how we use the PID which determines whether the PID is valid or not (e.g. whether we ever supply ff).

ktskt, 2019-01-04:
@peterbarker I think protection codes should care about non-negative values. 
However, it would be much better if we have "is_valid(...)" call as described in the parameter document.
Here is the document link http://ardupilot.org/copter/docs/parameters.html

More importantly, ArduCopter cannot protect against my cases *even* with the "is_valid(...)" function. 
Based on my understanding, that code is in "parameter_checks" that is called by "pre_arm_checks" function. 
It may check parameters only when it tries to switch to ARM mode. 
However, such parameters can be updated even after a drone is ARMed (e.g., GCS command). 
Therefore, I think that codes you suggested cannot protect against my example cases. 
That's why my examples cause such extremely unstable states.

rmackay9, 2019-01-04:
negative numbers for the PID values is definitely bonkers so protecting against that would be good.

ChrisBird, 2019-01-05:
Hi,
Yes I'd agree that negative PID values wont make sense not sure I didnt think of that when I did the initial change.  They really should be is_positive's instead (assuming 0 isnt counted as a positive).

On the part about checking on use or on loading it I'd suggest the current check at as part of arming is sufficient on the vehicle side with the exception of negative values.  That could be added without much overhead on the vehicle side and with an easy fall back position.  

Remember people may have vastly different vehicle types and I dont think we should be limiting it if they really want to set it to odd values (what is odd to us may not be odd to them)..

@ktskt if you want an easy one (just changing the value to is positive and not zero) to start with this is it.  If not I'm happy to do that sometime this week.  It would be easy enough to send a reject on PARM_SET call if it failed validation.

ktskt, 2019-01-06:
@ChrisBird If you plan to do that, I appreciate your efforts since I may not be able to patch that bug this week.

ChrisBird, 2019-01-06:
Hi @ktskt 
No probs, I'll put it on the list for this week.

ChrisBird, 2019-01-14:
Closed my pr, peter has a better solution.

ktskt, 2019-01-30:
I think it is fixed. So, I close it.

closed 2019-01-30 by ktskt