ArduPilot/ardupilot #10148: LOG_DISARMED options

Naterater, 2018-12-31:
### Feature request

Problem diagnosis for new users and even advanced users can sometimes be difficult (LOG_DISARMED is set to 0 by default, so in-air disarms do not contain valuable information). I think this feature could be expanded on intelligently. I also think the default behavior should include logging while disarmed, but probably at a reduced rate so we aren't wasting memory on SD cards.

**Describe the solution you'd like**
More options for LOG_DISARMED
0. Disabled
1. Enabled always
2. Enabled at a reduced rate (maybe 1 or 5Hz)
3. Enabled after first arming

Another parameter could be created (maybe LOG_EMPTY_DELETE) that would automatically clear any logs where the vehicle was never armed. 

**Describe alternatives you've considered**
Larger memory cards to deal with the additional memory

**Platform**
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
Recent disarming in-flight events https://github.com/ArduPilot/ardupilot/issues/10139

proficnc, 2018-12-31:
X3

hdtechk, 2019-01-01:
X4 or allow logging up to a set time limit after disarm

Pedals2Paddles, 2019-01-01:
^^ That would be the option I select hands down!

Naterater, 2019-01-01:
Edited above to include your great ideas.