ArduPilot/ardupilot #10152: Plane: (Quadplane) Transition does not finish when changing from any Q_ mode to FBWA/B

ignaciobfp, 2019-01-01:
### Bug report

**Issue details**

When transitioning to FWBA/B on a quadplane coming from any Q mode, the motors do not go down to their expected position in plane. Instead, the angle from vertical follows the throttle stick input, to go back up when throttle is down again. This is seen on a tricopter with front 2 vectored motors, configured as quadplane.

This model lacks airspeed sensor. Transition from manual plane modes to FWBA/B is correct. We are unsure if this bug was present on latest stable, or if it is related to the GPS speed not being high enough, although we tried running with the model with no luck (Q_ASSIST_SPEED is set to 0 to avoid interfering).

Bug reproduced by @IamPete1 on SITL on issue #10039

**Version**
Bug seen on: master (built in: 2018/12/29 , probably present much before)

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quadplane, 3 motors. Front 2 are vectored for yaw correction on copter modes. all 3 motors are tilted to horizontal position while in plane modes.

**Hardware type**
Pixhawk (FMUv2)

**Logs**
Not saved. Will try to get a log showing this problem ASAP.

IamPete1, 2019-01-01:
I'm not 100% sure this is a bug, I think it may be expected behavior. Have you seen any change to the behavior since earlier versions? 

With Q_ASSIST_SPEED at zero the issue should be removed when switching into FBWA from maunal mode and arming into FBWA, as the transition code is not used. 

Then when switching to FBWA from a Qmode you should see this behavior until you reach ARSPD_FBW_MIN, once this speed has been reached the vehicle should act just like standard FBWA.

ignaciobfp, 2019-01-02:
I am not completely sure it is a bug either, I wanted to perform some tests this morning to confirm this happens also when the plane is moving. I could not get the plane to do the transition, it does not matter if i lifted it or ran with it while appliying throttle, that whats makes me thing something could be wrong with the transition. Maybe the fact of not having airspeed sensor is also a problem, but my view is that transition should end aswell at some point, of course with worse handling/transition. Maybe speed on a person running was not enough to trigger this? If that is the case it should be configurable.

What you say is correct, with Q_ASSIST_SPEED 0 this issue is not present when coming from manual mode or arming directly in FBWA.

I wasn't able to test previous versions to check if this happens. If i remember correctly, i think it did not happen on stable, but I will confirm it this weekend with more time.

IamPete1, 2019-01-02:
> Maybe speed on a person running was not enough to trigger this? 

The threshold speed is set by ARSPD_FBW_MIN, I think its quite likely you couldn't run fast enough if its still default, you could lower it to say 4 or 5 m/s and try again. I think throttle position shouldn't matter.

ignaciobfp, 2019-01-02:
It is possible, I didn't know that parameter also affected the transition when no airspeed sensor is present, will check on a few days.

IamPete1, 2019-01-25:
@ignaciobfp any update on this? I will close for the time being, we can always reopen if you can confirm the bug. Thanks

closed 2019-01-25 by IamPete1

ifanirmala, 2023-07-22:
hello, I have a problem. My plane is a tricopter plane with 2 front tilt servos. When I use QSTABILIZE mode, the throttle can be maximum and can lift the plane. whereas when changing to FBWA plane throttle motor mode it feels like there is a limit and it can't be maximized, is there a solution so that the throttle in QSTABILIZE and FBWA modes can be the same maximum?