ArduPilot/ardupilot #10158: GCS_MAVLink: commit 6e7b9aa breaks message output to OSD panel

vierfuffzig, 2019-01-02:
### Bug report

@utkinpol noticed onboard OSD message panel not working as in prior firmwares. i could reproduce that OSD_MESSAGE does not show any messages anymore unless a mavlink telemetry connection is established.
a bisection search exposed https://github.com/ArduPilot/ardupilot/commit/6e7b9aa to break OSD_MESSAGE display without active telemetry connection.

**Platform**
[  ] All
[  ] AntennaTracker
[ x] Copter
[ x] Plane
[  ] Rover
[  ] Submarine

auturgy, 2019-01-02:
> a bisection search exposed 6e7b9aa to break OSD_MESSAGE display without active telemetry connection.

@peterbarker

vierfuffzig, 2019-01-02:
i now realize it's a number of changes in between last working and first broken version that i didn't bisect yet...

magicrub, 2019-01-02:
@vierfuffzig Is this OK to close?

vierfuffzig, 2019-01-03:
@magicrub i personally would prefer to get it solved as imho many onboard OSD users (including myself) do rely on that message panel pretty much. 
telling at a single glance what exact part of the GCS commits broke the message output is well beyond my scope though, and it‘s getting too late for me to do the required per-commit bisection tonight. that doesn‘t necessarily require to keep this actual issue open, but it likely needs a fix of some sort anyways?

peterbarker, 2019-01-03:
@vierfuffzig I believe https://github.com/ArduPilot/ardupilot/pull/10160 should fix the problem.

peterbarker, 2019-01-03:
On Wed, 2 Jan 2019, vierfuffzig wrote:

> @magicrub i personally would prefer to get it solved as imho many onboard

Please check the latest master.

vierfuffzig, 2019-01-03:
tested working. thanks a lot @peterbarker !

closed 2019-01-03 by vierfuffzig