ArduPilot/ardupilot #10173: Rover crash check action reliability

warp2, 2019-01-04:
Feature request

Description of the problem
At the moment in rover there is only Hold and HoldAndDisarm for crash check action.
I think if we would have two different crash check action it would enhance the reliability of long missions for rover platforms.

Solution
The idea would be to have two separate crash check actions:
1) As it is (if the rover flips over or tilt more than "CRASH_ANGLE" the rover would go to Hold or HoldAndDisarm as today)
2) If the rover would simply hit something or get entangled into something the rover would go on hold a configurable delay time and then try to complete the mission or repeats the delay-hold-auto cycle a configurable number of times before it goes to Hold or HoldAndDisarm as today.

Especially for boats it is very likely that on a river or lake the boat will hit a limb. If the boat would get into hold mode for a certain time and drift off there is the chance it could get free and could continue the mission or repeat the hold-auto cycle a number of times before completing the mission. If it get not free in the set number of hold-auto cycles it would go permanently to Hold or HoldAndDisarm.

Relevant Tag List:
Failsafe

Platform
[ ] All
[ ] AntennaTracker
[ ] Copter
[ ] Plane
[x] Rover
[ ] Submarine

Additional context
If people are generally OK with this, I am willing to test any implementation.

khancyr, 2019-01-04:
Hello,

This is interesting. A much complete approach would be to implement recovering behavior with delay : 
turn on spot
try to get backward
etc.
We could look at those that are implemented in ROS navigation, they handle a bunch of case already that we could reimplement here.

I would love to see those !

warp2, 2019-01-08:
Hello khancyr
I agree with you that a more complete approach would help even more.
Out of experience I think the reverse 2m and then go to hold would already solve a lot of problems.