ArduPilot/ardupilot #10175: Do not overwrite last WP location when switching to ATUO

Naterater, 2019-01-04:
### Feature request

**Is your feature request related to a problem? Please describe.**
When I conduct surveys, failsafes or other events may cause a deviation from a flight line. Upon continuing the mission, the flight-line is disregarded and the aircraft heads directly to the waypoint that it was tracking before. To properly cover the flight line, the whole line must be re-flown starting at the previous waypoint.

**Describe the solution you'd like**
Have a parameter that allows the navigation controller to keep the previous waypoint when resuming a mission. The aircraft will then navigate to the flight line upon entry into automatic mode.

**Describe alternatives you've considered**
Restart the whole flight line at the previous waypoint.

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[X] Plane
[X] Rover
[  ] Submarine

**Additional context**
N/A

IamPete1, 2021-07-26:
Fixed by https://github.com/ArduPilot/ardupilot/pull/13208 just add a `DO_SET_RESUME_REPEAT_DIST` WP to your mission

rmackay9, 2021-07-27:
I'm going to re-open this one if that's OK.  I don't think the repeat dist thing is a really easy solution for the user.

IamPete1, 2021-08-29:
@rmackay9 How can we improve? You think the functionality is OK but the interface is too hard? If I recall a dedicated param was voted down in #13208

rmackay9, 2021-08-30:
@IamPete1,

Ideally we should store the origin and destination of the line we are on and then use it during a resume to calculate the closest point on the line and then start from there on our way to the next waypoint.  A possible optimisation might be to record the current and previous nav command index and then recalculate the origin and destination during the resume but handling the terrain altitudes could be a bit tricky.

rmackay9, 2021-08-31:
@IamPete1,

I think what the user wants is clear and it's just that two solutions have been proposed (only one of them has been implemented).  I think my proposed solution could replace the existing solution.

IamPete1, 2021-09-12:
@rmackay9 Your proposed solution results in skipping part of the mission leg, the goal of  `DO_SET_RESUME_REPEAT_DIST` was to ensue all of the mission is covered for the mapping use case. As I say its the difference between "get back on track as soon as possible" and "go back to where we were and carry on from there".

I think this issue is resolved by `DO_SET_RESUME_REPEAT_DIST`.

We have this issue to track "get back on track as soon as possible" https://github.com/ArduPilot/ardupilot/issues/15557

closed 2021-09-12 by IamPete1

rmackay9, 2021-09-13:
@IamPete1 in general we try to avoid a little tug-of-war over what issues are open/closed.  In any case, I've updated the other issue to make it apply to all vehicles.  I really don't think that the do-set-resume-repeat-dist is exactly what the users want.